Ultimaker/Cura #11129: program not slicing to gcode.

mrplasticcup, 2021-12-24:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Creality ender 3

### Reproduction steps

Tried to save.stl file and slive to g code.  program stated there was an error.

### Actual results

Nothing happened and there seemes to be something wrong with the file.

### Expected results

File should have been saved to gcode.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Vandresc, 2021-12-27:
Hi @mrplasticcup, could you please share a project file so we can try to reproduce it? Thank you!
File>Save project (you will probably need to zip the file first)

mrplasticcup, 2021-12-28:
Small file.  Saved project from Cura.

rburema, 2021-12-30:
Hello @mrplasticcup 
Attaching via email doesn't work (sadly) you need to attach the file via the github website itself.

closed 2022-01-17 by no-response[bot]