Ultimaker/Cura #11130: Temp Tower does not load (or build) correctly

cgibke, 2021-12-25:
### Application Version

4.12.0

### Platform

Windows 7

### Printer

Creality Ender 3 pro

### Reproduction steps

1. Notified of new version of Cura, have been using v4.11
2. Downloaded and installed new version, uninstalled old version via new install procedure
3. Added calibration plugin
4. restarted Cura
5. added PLA Temp Tower to build plate
6. added calibration script, set for 41 change level, 4 base
7. changed profile to CHEP 4.0_12 (Good)
8. set temperature and starting temperature to 220
9. set retraction distance to 2 for all-metal hotend
10. clicked "slice"
11. selected "Preview"

### Actual results

Object in "Preview" does not include side columns (looks weird
[TempTower4_12.zip](https://github.com/Ultimaker/Cura/files/7775841/TempTower4_12.zip)
)

### Expected results

correct tower model as compared to what Cura v4.11 produces

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

screen shot included
Cura 4.11 displays this object correctly

GregValiant, 2021-12-26:
Change your "Outer Wall Inset" to 0.  The walls are so thin that there is an inside outer wall and an outside outer wall.  You are moving both of them in towards the middle of the wall and it leaves nothing to print at .4 line width.
Your Outer Wall Wipe Distance is too high as well.  Drop that down to .1 or .2.

cgibke, 2021-12-26:
Actually, I didn’t change those values – I installed the program and a third-party profile. The same combination worked fine in v4.11 via the same procedure.

Thanks for the reply, though.

…….cg

cgibke, 2021-12-26:
As it turns out, the third-party profile I used was what killed the build. Using the standard profile, it builds correctly.

I will attempt to contact the creator of the profile (CHEP - Chuck Hellebuyck's Electronic Projects) and advise him of the problem.

Thanks

And Merry Christmas!

…….CG

GregValiant, 2021-12-26:
Chuck puts together pretty good stuff.  I've watched a few of his videos and I like them.
The Outer Wall Inset can be used to adjust dimensional problems but most printers and models don't really need it.  On a thin wall model like you have, it can (and did) cause problems.
People like to talk about profiles.  I have 3 (1 each for PLA, PETG, TPU) and they are the same except for temperatures and speed.
Next to the Settings Search box is a drop down list for Setting Visibility.  I always set it to "All" and for every model I go down through the whole list and make sure each setting is what I want for that particular model.  There is no "one-size-fits-all" profile.  I make my changes and then stare at the preview.  It takes some time, but when I'm done I am confident that my print will be successful and be what I want.

rburema, 2021-12-30:
@cgibke 

> As it turns out, the third-party profile I used was what killed the build. Using the standard profile, it builds correctly.

I will be closing this then. If there is (another) issue, please don't hesitate to report it.

@GregValiant Thanks for the help, as always.

closed 2021-12-30 by rburema