Ultimaker/Cura #11132: Slicing failed with unexpected error.

PernelSned, 2021-12-25:
### Application Version

4.12.1

### Platform

Windows 11

### Printer

Custom FFF

### Reproduction steps

1. Import .STL
2. Slice Model

### Actual results

Slicing failed with unexpected error.

### Expected results

Gcode.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7776348/cura.log)
[CFFFP_Jim_Solder_Fingers_Base.zip](https://github.com/Ultimaker/Cura/files/7776362/CFFFP_Jim_Solder_Fingers_Base.zip)
[image]

rburema, 2021-12-30:
Thanks for the report, I can confirm this happens in 4.12.x, 4.13-BETA. and even an early preview of 5.0/Arachne (beta 2).

The model itself is proper (not broken and has correct normals), so it can't be that.

I'll add it to the list of things to discuss with colleagues.

fvrmr, 2022-01-04:
We have discussed it and created a ticket on our backlog for this bug.
Devs see CURA-8861

nallath, 2022-01-04:
The odd thing is, I can't reproduce this on the beta or on the latest 4.13 on Linux...

nallath, 2022-01-07:
I've managed to reproduce this on windows. It turned out that you had set the max retraction count to a value that caused the crash. If you change it to something lower (eg: 999999999) it will work. 

In the next release, the interface should also prevent one from entering too large numbers. Thanks for reporting it!

commit ecdf536d (2022-01-07): Add max value to max_retraction_count

CURA-8861
Fixes #11132
commit 25ec5887 (2022-01-07): Add max value to max_retraction_count

CURA-8861
Fixes #11132
closed 2022-01-07 by nallath