Ultimaker/Cura #11133: Cura generates invalid G-Code.

rpweaverjr, 2021-12-26:
### Application Version

4.12.1

### Platform

Dell XPS 1645 running Windows 10 Pro 21H1

### Printer

Ender 3 Pro Running Marlin

### Reproduction steps

Created a G-Code from an STL file created in Fusion 360. At about 95%, the printhead leaves the object being printed and transits to the right side of the build plate and then starts printing in thin air. This happened twice with different although evolved STL files. I tried to include the STL file but that was rejected

### Actual results

The printhead extrudes plastic into the air and the print fails.

### Expected results

The printhead should not have left the object being printed.
This is an hour and a half print so cannot easily reproduce without a lot of wasted time. When I tried to Preview, Cura crashed.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

I can supply the STL if you need it
[CenterHub3.zip](https://github.com/Ultimaker/Cura/files/7776512/CenterHub3.zip)
.

GregValiant, 2021-12-26:
A 3mf project file contains the model.
Cura can open Gcode files.  The Gcode reader is separate from the slicer so you will get an accurate preview in Cura.
I sliced your model with no changes and there were no problems in the gcode.  I read the gcode into AutoCad to take Cura out of the loop (screenshot is below).  If you could zip and post the problem gcode file it may show something.
You can see in the image that there are no odd movements away from the part.  It goes from the purge lines to the skirt, prints the model, then raises and does the wipeout before moving to the left rear corner.

[image]

rpweaverjr, 2021-12-26:
This is the part being printed in the video.

Sent from my Verizon, Samsung Galaxy smartphone

[01]<https://user-images.githubusercontent.com/64202104/147408503-372955c1-3888-45c3-94f1-d76a64dac93d.jpg>

rpweaverjr, 2021-12-26:
I trimmed just the offending part. THIS ONE IS EVEN WORSE. Creality Slicer prints it properly.I have a video but it is too large, 15MB. How can I send it?

Sent from my Verizon, Samsung Galaxy smartphone

rpweaverjr, 2021-12-26:
How do I attach a video? I select one,  tap DONE, and nothing happens.

rpweaverjr, 2021-12-26:
Actually the cider is just 5MB

rpweaverjr, 2021-12-26:
Here is the part on the Cura screen.
[image]

GregValiant, 2021-12-26:
Put the problem Gcode file into a zip file and post the zip file here.  My printer is an Ender 3 Pro and I have some old filament laying around and so I could try to print it if I had to but the gcode will tell the tale.
All that can be told from a video is "something ain't right".

rpweaverjr, 2021-12-26:
[TopRing.zip](https://github.com/Ultimaker/Cura/files/7777368/TopRing.zip)
This is the GCode for the top portion in the video and the screenshot.

rpweaverjr, 2021-12-26:
Cura says it takes 18 minutes to print. The problem starts at around 3:20 or so depending on how long it takes to heat things up.

GregValiant, 2021-12-26:
That's the STL file.

rpweaverjr, 2021-12-27:
Sorry
[TopRing.zip](https://github.com/Ultimaker/Cura/files/7777744/TopRing.zip)
.

GregValiant, 2021-12-27:
That gcode file is a complete mess and is not even close to being correct.  Missing layers, support moved to layer 36?

The LAYER COUNT is 27 but there are only 10 layers in the gcode file.  The layers are:
Layer:0 at Z=.28
Layer:11 to Layer:17 at Z=1.44 to Z=2.16
Layer:35 @4.32
and finally:
;LAYER:36
;TYPE:SUPPORT
The "End Gcode" section is missing from the file so nothing turns off at the end.  The line "End of Gcode" line as well as the entire "Settings" section that should be at the very end of the file are also not present.
The file ends abruptly with
G1 X106.58 Y121.073 E856.0832
G0 F9000 X107.036 Y120.811
G1 F1500 X1

I would think there is a problem with the SD card but then there is this from the end of LAYER:17...

G1 F1500 X101.795 Y101.512 **E819.62269**
G0 F9000 X101.813 Y101.532
G1 F1500 E813.12269
;MESH:NONMESH
G0 F300 X101.813 Y101.532 Z4.32
G0 F9000 X101.306 Y101.498
G0 X97.994 Y98.364
;TIME_ELAPSED:2340.391844
;LAYER:35
;TYPE:SUPPORT
G1 F1500 **E819.62269**
G1 X98.526 Y97.839 E819.6376
So the E value at the end of Layer 17 is carried over correctly to the start of Layer 35.

Here is a screenshot of the gcode opened in Cura.  It looks the same when read into AutoCad.

[image]

GregValiant, 2021-12-27:
@fvrmr,
I'm including one of my own gcode files that appears to have problems with the Layer Count.  Note the line near the beginning:
;LAYER_COUNT:175
but the top layer in the file is
;LAYER:119

My gcode file printed correctly.

I have a post processor active that adds M117 and M118 lines at each layer.  Note that at the end of the file they get added back-to-back without a line ";LAYER:" present.  That has never happened before.
If you need a project file I will have to see if I can put one together.  I may have over-written the STL file.

rpweaverjr, 2021-12-27:
The SD card is almost full but I would expect the write to be rejected,  not corrupted. Did you try slicing the STL file yourself? I will clear out some space and see what happens. You have to admit that Cura did something wrong, whether it is creating a bad gcode or failing to reject a full SD card. This would also verify that the STL file is not corrupt although Cura did display it correctly.

Sent from my Verizon, Samsung Galaxy smartphone

[image]<https://user-images.githubusercontent.com/64202104/147476344-ef068f0a-6d13-4e49-aa15-a86c975a9df7.png>

rpweaverjr, 2021-12-27:
Windows says I have 35MB free on the card.

Sent from my Verizon, Samsung Galaxy smartphone

rpweaverjr, 2021-12-28:
The latest one was sliced with Creality Slicer (based on Cura) and it is also messed up. Cura 4.12.1 gets stuck trying to slice it. But it succeeded the second try and the Preview looks good this time. I won't know until I try to print it. It almost seems like the STL file has garbage in it because I moved things around and made copies and all other sorts of manipulations in Fusion 360. It seems like some of these operations are ghosted in the gcode.
[FullSplit.zip](https://github.com/Ultimaker/Cura/files/7782770/FullSplit.zip)
.

rpweaverjr, 2021-12-28:
Cura 4.12.1 crashes every time I close it. I have a feeling the gcode is not completed. It seems to hold the removable drive until I close Cura. I cannot Eject the removable drive until then.

fieldOfView, 2021-12-28:
Can you try saving the gcode to your harddrive instead of using the "Save to removable drive" option? Then use explorer to put the file on your removable drive instead.

GregValiant, 2021-12-28:
That latest gcode file in FulSplit.ZIP doesn't really look like it was sliced with Cura 4.12.1.  The opening header is a lot different than what I get.
Mine:
;FLAVOR:Marlin
;TIME:1714
;Filament used: 1.71283m
;Layer height: 0.2
;MINX:84.401
;MINY:80.688
;MINZ:0.2
;MAXX:145.502
;MAXY:149.218
;MAXZ:7.6
;**Generated with Cura_SteamEngine 4.12.1**

Yours:
M190 S105.000000
M109 S255.000000
;Sliced at: Mon 27-12-2021 18:08:30
;Basic settings: Layer height: 0.12 Walls: 0.8 Fill: 50
;Print time: 45 minutes
;Filament used: 1.792m 5.0g
;Filament cost: None
;M190 S105 ;Uncomment to add your own bed temperature line
;M109 S255 ;Uncomment to add your own temperature line
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops
G1 Z15.0 F3000 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F3000
;Put printing message on LCD screen

rpweaverjr, 2021-12-28:
The latest gcode is from Creality Slicer (Cura based). I included it because it messed up also. I left that message in Git hub. It may point to a design/implenentation flaw that is very old.
The latest geode that I made last night from Cura was perfect. This is really strange.  It is almost like an array is dynamically allocated and the garbage data was used without clearing it first. That is the only way I can explain the ghosting of older stl files.

Sent from my Verizon, Samsung Galaxy smartphone

rpweaverjr, 2021-12-28:
When I slice and print just the top section, it always messes up in the same place even though I make slight adjustments to the design. The last successful slice/print was when I did all three sections together (to avoid supports). I have not caught when it goes wrong. It just seems to start printing in mid air. The Preview looks good though
[Top.zip](https://github.com/Ultimaker/Cura/files/7785687/Top.zip)
.

rpweaverjr, 2021-12-28:
Well, I saw it this time. At layer 3 or 4, it lifts straight up about 1/2 inch and starts printing in thin air.  The previous one was stored to my hard drive. This was stored directly to the removable drive
[TopRemovable.zip](https://github.com/Ultimaker/Cura/files/7785743/TopRemovable.zip) They should be identical but they aren't. The preview of the Removable version clearly shows the lift and some other thing off to the right. And the top profile is not what is in the Design. As you suggested, I will copy the hard drive version to the removable and see what happens.
.

rpweaverjr, 2021-12-28:
Copying from the hard drive to the removable drive seems to have fixed the problem. Whether it is a Cura problem or a Windows problem when writing directly to the Removable drive is an open issue. It looks to me like some previously used and deleted file allocation blocks still have old gcode in them and it is not being overwritten. This then gets incorporated into the gcode and makes a mess.

GregValiant, 2021-12-28:
Did you try @fieldOfView 's advice?  Have you tried using a different SD card?  Your printer doesn't deal with cards over 32gb.

rpweaverjr, 2021-12-28:
The card I am using is only 512MB so size shouldn't be a problem.

Sent from my Verizon, Samsung Galaxy smartphone

GregValiant, 2021-12-29:
On the hardware side - are you using an SD to mini-SD converter?  I use one and it works well but some of them are pretty cheesy.

rburema, 2021-12-30:
> The card I am using is only 512MB so size shouldn't be a problem.

Regardless, could you try and save to the hard-disk and see if that makes any difference?

The preview being _completely_ different from the actual gcode-file is a big hint that it might not be Cura.

rpweaverjr, 2021-12-30:
I did that and it worked fine. I used the same SD card too.  I already reported that. I'm not sure if it was on Github or email.

Sent from my Verizon, Samsung Galaxy smartphone

Ghostkeeper, 2022-01-03:
> The latest geode that I made last night from Cura was perfect. This is really strange.  It is almost like an array is dynamically allocated and the garbage data was used without clearing it first. That is the only way I can explain the ghosting of older stl files.

This is a strong indicator of broken SD cards, indeed.

The issue with a broken SD card is that Windows doesn't give Cura any indication that the file write might have failed. Cura calls on Python to `open("filename.gcode").write("<gcode>"), and sees no exceptions, so it thinks the write has completed. Indeed, Windows itself doesn't even know that an SD card might be broken in many cases, unless the user [explicitly scans it for broken sectors](https://www.diskpart.com/windows-10/bad-sector-repair-windows-10.html).

As such, there is hardly anything that Cura can do to prevent this issue. I think you'll just have to replace that SD card, or try wiping it (at risk of it being a hardware issue).
And be sure to eject the SD card before removing it from the computer, because failing to do so can cause issues like this.

closed 2022-01-03 by Ghostkeeper

rpweaverjr, 2022-01-03:
Funny thing is that I had Windows scan the card and found no problems. When I had Cura write to the hard drive and used Windows to copy the file to the same exact SD card,  the file was perfect and printed as expected. I overwrote the previous corrupted file on the SD card. I can't explain this if the SD card is bad.

Sent from my Verizon, Samsung Galaxy smartphone

Closed #11133<https://github.com/Ultimaker/Cura/issues/11133>.

rpweaverjr, 2022-01-04:
I just had another repeat of the erroneous printer activity so I will concede that the SD card must be bad. This time I wrote to the hard drive from Cura and copied the gcode to the SD card with windows.

Sent from my Verizon, Samsung Galaxy smartphone