Ultimaker/Cura #11136: Tree support - doesn't honor support bottom distance

mslucass, 2021-12-27:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

Ender 3 v2 - BTT MKS-2 motherboard - Marlin firmware

### Reproduction steps

slice with different settings

project: with normal support does have an empty line at print line 10
The project with tree support doesn''t have this =>  result melted support against the main print

### Actual results

Print line 10 should have a empty line but with thee support does print something. result difficult to remove

### Expected results

an empty print line

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura 4.12.0.zip](https://github.com/Ultimaker/Cura/files/7779372/Cura.4.12.0.zip)

Ghostkeeper, 2022-01-03:
Indeed, I can reproduce this again. We [fixed this at some point](https://github.com/Ultimaker/CuraEngine/pull/1288/files), but it seems to have regressed.

fvrmr, 2022-01-07:
We have created a ticket on our backlog for this.
Devs see CURA-8874

mslucass, 2023-05-15:
> Hey @mslucass<https://github.com/mslucass>,
>
> Sorry for the late reply.
> We are cleaning up our documentation.
>
> I found your issue and noticed that it's related to tree supports.
> The upcoming stable release Cura 5.4 will have big improvements regarding tree support and support interface.
> It's based on the code contributed by Thomas Rahm, you can learn more about it here: #14111<https://github.com/Ultimaker/Cura/discussions/14111>
>
> I took a look at your file in a nightly build, and the new behavior looks like this.
> [image]<https://user-images.githubusercontent.com/40423138/238403981-023711cb-685b-496f-a15a-2a8113efdbd6.png>
>
> Can you let us know if that is the behavior you are looking for?
> Because to me, it doesn't seem like a bug in Ultimaker Cura anymore.
> Would you still like to keep this issue open or do you agree that we can close it?

You can close this issue

Maurice

MariMakes, 2023-05-19:
Thanks for the update 😊
I'll close the issue

closed 2023-05-19 by MariMakes