Ultimaker/Cura #11137: Thumbnail not shown (win10)

GMagician, 2021-12-27:
### Application Version

4.13.0 beta

### Platform

Windows 10

### Printer

Bear 2.1(Custom)

### Reproduction steps

Save a .3mf file with 4.13 version, it adds a jpg file into .3mf's "metadata" folder
Open explore on folder containing a .3mf file

### Actual results

Thumbnail is not shown, windows displays Cura icon instead

### Expected results

Thumbnail image shown

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Not sure this is a "cura" issue but more a window integration issue (I don't even know if it is your relevance)
I started with windows without .3mf association, double click on icon open windows association window, cura is installed, such file has no associated icon 
Uninstalled cura and reinstalled (checking .3mf file support).
Now when I first double click on icon, windows suggest cura as application. I select it. And now a cura icon is shown on 3mf files.

I see that cura creates a 3mf file association tied to its version (key: HKLM\Software\Classes\.3mf.4.13).
To properly show thumbnails it miss something. under .3mf.4.13 it must be added a "shellex\{e357fccd-a995-4576-b01f-234630154e96}" subkey and its "default" content should be "{E64164EB-1AE0-4C50-BAEF-A413C2B3A4BC}"

Once this is done, thumbnails are properly shown

Vandresc, 2021-12-27:
Hi @GMagician, this is fixed now in the stable 4.13. Thank you for your feedback!

closed 2021-12-27 by Vandresc

GMagician, 2021-12-29:
I also noticed that sometime jpeg/metadata folder file is not added to saved project.

Ghostkeeper, 2022-01-03:
Do you have reproduction steps for that?

GMagician, 2022-01-03:
I have not found any "sequence". Sometime, after 4/5 times I save the same project, thumbnail appears