Ultimaker/Cura #11138: Autoscale

John1900, 2021-12-27:
### Is your feature request related to a problem?

Building models from early engineering drawings in inches requires rescaling to metric on every model loaded to Cura.  Cura used to do this automatically with "automatically scale small models".  Does not work with current release.  a Barbarian Units add-in is available but has to be run on every load.

### Describe the solution you'd like

A setting for "Default Scale".   For most users this would be left at 1.  Others could set this to 2540 or whatever is needed for scaling their drawings to Metric.  

### Describe alternatives you've considered

Going back to "rescale small objects" is problematic as what is "small".  And in any case some other scale factor may be needed.

The existing scaling requires application on every model load.

### Affected users and/or printers

Anyone working with Imperial units or from drawings that need to be scaled before printing

GregValiant, 2021-12-28:
One thing you can try now is to load the "Mesh Tools" from the Marketplace.  After restarting Cura - under "Extensions | Mesh Tools | Mesh Tool Settings" there is a drop down for "Units for files that don't specify a unit" and at the bottom you can choose Inches or Feet.  My projects are about half and half so whether I set it to Inches or Millimeter's it's only right 50% of the time.

fieldOfView, 2021-12-28:
> My projects are about half and half so whether I set it to Inches or Millimeter's it's only right 50% of the time.

I am not making any promises, but would you like an option to always ask what the unit is when opening an STL in Cura? I think I can add that in MeshTools.

GregValiant, 2021-12-28:
Hi @fieldOfView .  I wouldn't ask for it since I design either in inches or mm's.  I leave the scaling in mm's and if a model comes in very small I scale it up.  Since I am the designer and the slicer I know what I did (although at my age actually remembering is sometimes a problem).
The difference between a popup and just going over and changing the scale manually would only be one or two clicks.  I'm pretty sure some people would prefer it though.

John1900, 2021-12-28:
Mesh tools work fine.  The key for me is that the scaling is automatic on
load.

Thanks very much.

Ghostkeeper, 2022-01-03:
> Cura used to do this automatically with "automatically scale small models". Does not work with current release.

As far as I can see it still works. It only applies when the model is less than 1/100th of the size of the build plate in all dimensions though. As a result, it works well for millimetres vs. metres, but not so well for millimetres vs. inches. Making it more sensitive would indeed be problematic, because people regularly print things that are quite small. There's definitely more than a factor 25.4 difference in the range of model sizes people print.

There is something to say for building this in Cura by default, but since both the Mesh Tools and the Barbarian Units plug-ins implement this, it's not going to get a very high priority for us.

John1900, 2022-01-03:
No problem. Mesh tools works fine and is automatic.

Perhaps some “advertising” would help.  I struggled for months manually
converting on every build.  Why on earth would I ever try something called
Mesh Tools to do a units conversion?   Perhaps something in the help under
units conversion or scaling would do the trick.

Thanks for the support.

John

Ghostkeeper, 2022-01-04:
Well, presumably you'd know that your units are barbaric and to use the Barbarian Units plug-in :stuck_out_tongue: There are also different file formats, like 3MF, that specify that their units must be millimetres so anything that writes 3MF will write it correctly. This sort of thing is the future I think.

Ultimaker does promote some plug-ins as "featured", but there's limited space there. I guess Barbarian Units didn't make it, but Mesh Tools did.

fieldOfView, 2022-01-04:
Hi, author of the MeshTools plugin here. The MeshTools plugin is a "swiss army knife" type of tool that combines a lot of useful functionality. See the readme for details: https://github.com/fieldOfView/Cura-MeshTools/#readme

The scaling on load that is performed by the MeshTools is a bit different than how the Barbarian plugin does it; with the MeshTools plugin, the actual coordinates of the mesh are scaled up. This way the model in Cura is at 100% scale, whereas with the Barbarian plugin it will be at 25400% after applying the scale. I did it that way because it is easier to work with if you want to do further scaling (eg converting from inches AND making the model one and a half times as big, you end up with a scale of 38100% with the Barbarian plugin, and simply 150% with MeshTools).

closed 2023-07-13 by github-actions[bot]