Ultimaker/Cura #11139: Version 4.12.1 says to monitor printer please connect printer and it is connected I have a Monoprice Mini Delta V2 I really need help

DENTOASTER, 2021-12-27:
### Application Version

4.12.1

### Platform

windows 10

### Printer

MP Mini Delta V2

### Reproduction steps

I plugged in the usb than added the file and thats what it says 

### Actual results

Nothing

### Expected results

Connected to the printer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

WONT PRINT ANYTHING

GregValiant, 2021-12-27:
Some USB cables are for charging only and don't have data wires.
With the printer plugged in to the USB port go to Windows Device Manager and check for "Ports (Com and LPT)".  If Windows is acknowledging the connection it will look similar to mine which is shown below.
It is far better to print from the SD card, or use Octoprint, Repetier Host, or PronterFace/Printrun.

[image]

fvrmr, 2022-01-03:
Hi @DENTOASTER thank you for your report. 
We don't support printing via USB anymore since our printers don't use USB printing. But you can check if the USB printing plugin is disabled. Go to Marketplace - Installed and scroll down to USB printing. If this is disabled you can enabled it. 
[image]

If this is enabled and still not working your cable could be the case like @GregValiant mentioned. Since we don't support USB printing it is indeed better to use SD card or Octoprint. If you want to use USB printing you can use Printrun. 

I will close this issue since we won't fix this issue. But feel free to reach out if you have questions or need more help.

closed 2022-01-03 by fvrmr