Ultimaker/Cura #11140: Overhanging Wall Acceleration and Speed with gradual change

RedPetroleum, 2021-12-28:
### Is your feature request related to a problem?

No but it would enhance print quality without slowing it down significantly. 

### Describe the solution you'd like

I would like like more options for the experimental feature _Overhanging Wall (Angle / Speed)_.
When optimizing a printer hardware and software to the point that you can push the maximum volumetric flow without ringing, part cooling becomes the biggest problem and overhangs limit the print speeds the most. The _Overhanging Wall_ feature helps a lot but lacks a bit in options, so here are my suggestions:

First would be to change the speed gradually from a _Start Overhanging Wall Angle_ to a _Max Overhanging Wall Angle_ with the _Overhanging Wall Speed_. This would just make a lot of sense because a 50° overhanging wall can be challenging and could benefit from a slow down but 60° even more. 

Secondly, I would like to have Overhanging Wall Acceleration. 
Overhanging walls are not nearly as critically as overhanging sharp edges, because they tend to warp a lot and won't be taken into account by _Overhanging Wall Angle_ enough. Having a low print acceleration would help a lot for those overhanging edges. It would probably also make sense to have the option to change this gradually. 

For example:
`Start Overhanging Wall Angle = 30°`
`Max Overhanging Wall Angle = 70°`
`Overhanging Wall Speed = 50%`
`Overhanging Wall Acceleration = 30%`

→ any straight to 30° overhang wall would print with default wall speed and acceleration 
→ any wall that's overhanging more than 70° would print with 50% speed and 30% acceleration
→ a 50° wall would print with 75% speed and 65% acceleration

### Describe alternatives you've considered

Changing geometry or slowing down the entire outer wall speed and acceleration even though only a small part has overhangs. 

### Affected users and/or printers

This will affect people that like to optimize print time or everyone that uses a 0.6 or greater nozzle with a capable CoreXY machine. Basically all uses that reach high volumetric flow or have problems with cooling.

fvrmr, 2022-01-04:
Hi @RedPetroleum thank you for your feature request.
I will bring this up for discussion. Keep you posted!

ZenithAstralis, 2022-06-22:
I would Really like this feature~!

dedge10, 2024-08-18:
Would love this to as Creality Print offers this and is gradual, it is under speed 
[image]