Ultimaker/Cura #11141: Error slicing using trees

manjaroblack, 2021-12-28:
### Application Version

4.13 Arachne

### Platform

Ubuntu

### Printer

FLSun Super Racer

### Reproduction steps

Select Tree Supports. Slice.

### Actual results

Failure to slice

### Expected results

Slices fine in 4.13 Beta. It should also slice in Arachne

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7782593/cura.log)
[project.zip](https://github.com/Ultimaker/Cura/files/7782600/project.zip)

fvrmr, 2022-01-03:
Hi @manjaroblack thank you for your report.
I can reproduce your issue and even without support it fails slicing in Arachne. 
So the support is not causing this bug. But `connect infill lines` seems to be the issue. If you disable this you can slice correctly. 
I will raise this with the team. Keep you posted!

fvrmr, 2022-01-04:
I have discussed it with the team and I created a ticket on our backlog for this bug.
Devs see CURA-8866

closed 2025-07-20 by manjaroblack