Ultimaker/Cura #11142: Cura crashes on startup with Segmentation Fault

taDachs, 2021-12-28:
### Application Version

4.12.1

### Platform

Ubuntu 20.04

### Printer

Ender 5

### Reproduction steps

I start cura either from a snap install or from the appimage (Problem occurs with both options).

### Actual results

Cura crashes on startup. The loading window shows up before. Also no log is created.

### Expected results

Cura not crashing.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

No log is created, nor can I export a project file since the gui doesn't open. Starting cura in the console gives me the following output:

```
/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/snap/cura-slicer/58/usr/lib/python3/dist-packages/dbus/connection.py:337: DeprecationWarning: invalid escape sequence \
  """Arrange for the given function to be called when a signal matching
[MainThread] UM.Application.initialize [160]: Initializing Ultimaker Cura
[MainThread] UM.Application.initialize [161]: App Version 4.12.1
[MainThread] UM.Application.initialize [162]: Api Version 7.8.0
[MainThread] UM.Application.initialize [163]: Build type SNAP
[MainThread] UM.Resources.__initializeStoragePaths [447]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [457]: Config storage path is /home/max/snap/cura-slicer/58/.config/cura/4.12
[MainThread] UM.Resources.__initializeStoragePaths [465]: Data storage path is /home/max/snap/cura-slicer/58/.local/share/cura/4.12
[MainThread] UM.Resources.__initializeStoragePaths [477]: Cache storage path is /home/max/snap/cura-slicer/common/.cache/cura/4.12
[MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: /snap/cura-slicer/58/usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager.__init__ [55]: Found bundled packages JSON file: /snap/cura-slicer/58/usr/share/uranium/resources/bundled_packages/uranium.json
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [123]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [131]: Yay, we got at least OpenGL 4.1 core: 4.6 Core profile
[MainThread] UM.Qt.QtApplication.initialize [174]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [181]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [185]: Initializing version upgrade manager ...
[MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from /snap/cura-slicer/58/usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [170]: Loaded bundled packages data from /snap/cura-slicer/58/usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [197]: Loaded user packages management file from /home/max/snap/cura-slicer/58/.local/share/cura/4.12/packages.json
[MainThread] UM.PackageManager._saveManagementData [270]: Package management file /home/max/snap/cura-slicer/58/.local/share/cura/4.12/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [117]: Loading plugin configuration file '/home/max/snap/cura-slicer/58/.config/cura/4.12/plugins.json'
Cyclic dependency detected between "file:///snap/cura-slicer/58/usr/share/cura/resources/qml/Actions.qml" and "file:///snap/cura-slicer/58/usr/share/cura/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - fallback: "/snap/cura-slicer/58/usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Styles/Desktop"
/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py:316: DeprecationWarning: The icon 'search' is deprecated. Please use icon 'Magnifier' instead.
  self._qml_engine.load(self._main_qml)
Fatal Python error: Segmentation fault

Thread 0x00007f120b7fe700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 306 in wait
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/zeroconf/__init__.py", line 2165 in wait
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/zeroconf/__init__.py", line 1466 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
[... 81 lines cut ...]

Thread 0x00007f126affd700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f126b7fe700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Current thread 0x00007f1293d491c0 (most recent call first):
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 134 in beginRendering
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/Qt/Bindings/MainWindow.py", line 253 in _render
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 393 in exec_
  File "/snap/cura-slicer/58/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 871 in run
  File "/snap/cura-slicer/58/usr/bin/cura", line 233 in <module>
zsh: segmentation fault (core dumped)  cura-slicer.cura
```

fieldOfView, 2021-12-28:
Could you try the official AppImage instead of the snap? The snap is not maintained by Ultimaker.

taDachs, 2021-12-28:
> Could you try the official AppImage instead of the snap? The snap is not maintained by Ultimaker.

As mentioned in the issue, the problems occurs with both the snap and the app image. The error message is shorter tho:
```
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
Cyclic dependency detected between "file:///tmp/.mount_UltimapxEuvA/usr/bin/resources/qml/Actions.qml" and "file:///tmp/.mount_UltimapxEuvA/usr/bin/resources/qml/Actions.qml"
WARNING: Cannot find style "default" - fallback: "/tmp/.mount_UltimapxEuvA/usr/bin/qt/qml/QtQuick/Controls/Styles/Desktop"
Segmentation fault (core dumped)
```

fieldOfView, 2021-12-28:
> the problems occurs with both the snap and the app image.

Sorry, I missed that bit.

fvrmr, 2022-01-04:
Hi @taDachs thank you for your report.
I will bring this up for discussion. Keep you posted!

Ghostkeeper, 2022-01-04:
I'm unable to reproduce this so far, neither from source nor with the AppImage.

The cyclic dependency it finds also doesn't really make sense. It's the same file. And in [that file](https://github.com/Ultimaker/Cura/blob/master/resources/qml/Actions.qml) it doesn't refer to itself.

fvrmr, 2022-01-07:
I have discussed this with the team and nobody of the team could reproduce this issue. So we will have to leave this issue open and hope for more info about this. I hope you understand.

kamitor, 2023-04-06:
I have this issue too; 
Ubuntu 20.04 mate; Devault install
Steps to reproduce; Download ultimate cura app-image 3.10 or 3.11
Give rights and premissions trough right-click the application and make it an executable.
Start program, Program crashes. 
Start from Command line, Segmentation Fault.

MariMakes, 2023-10-19:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I was wondering if this is still an issue for you. 

Could you try out our latest version of Cura and let us know if it's resolved? 
You can download it here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

Thank you 🙏

taDachs, 2023-10-25:
I switched distros so I don't have the same setup anymore, but I can check in one of my dev containers