Ultimaker/Cura #11145: AnyCubic Mega S print via USB  & SD card failures

swdw973, 2021-12-28:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Anycubic Mega S

### Reproduction steps

1. Attempted to print via USB, 
2. Attempted printing the same file via SD card

### Actual results

Note: did not have this issue with Cura 4.11

With Cura 4.12.1- Attempting to print via USB has resulted in several failures. The program loses communication, the print head stops, but stays at temperature and sometimes keeps pushing filament resulting in a huge blob. The temperature reports are way off too. I actually have to reboot both the printer and shutdown / reload Cura to re-establish communication.

This is consistent with certain files failing, but the point on the object it fails may change but the height it fails at is consistent.

Exporting to an SD card will sometimes work. Other times it will never start to print. I can import the project to SuperSlicer , slice it and save the exact same project with the same settings to the SD card and it will print flawlessly.

The issue seems to be z axis related as I've had no problem printing flat objects through Cura via usb or SD card (max height 3.5 mm). With taller objects, it fails between 10 and 20 mm, depending on the object, or won't start at all from the SD card.

There are no log files being created from this. The fact it works with no issues in another slicer seems to indicate the stl and 3mf files are not the issue. 

### Expected results

Expected results? A finished print.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info

swdw973, 2021-12-29:
Reinstalled 4.11
All printing via usb problems gone with the numerous projects that were acting up. Same project that wouldn't start a print with the gcode saved to the SD card from 4.12, prints fine with the code from 4.11

Pretty much verifies this is a slicer related issue, not a printer issue.

Going to install the 4.13 beta and give it a try.

fvrmr, 2022-01-03:
Hi @swdw973 thank you for your report.
We don't support printing via USB since our printers doesn't use USB printing anymore. 
If you want to print via USB it is better to use Printrun. 
However could you share your project file (file - save project) of 4.11 and 4.12 in a .zip file here on Github? To check if this is indeed related to Cura.

Ghostkeeper, 2022-01-04:
Since our testers can start prints from SD cards fine, I reckon what we'd need to reproduce this is Swdw's SD card. Either that, or some information on what g-code your printer expects to see and what g-code Cura is producing instead.

If Cura's g-code output is wrong for this printer, we might be able to change the output. But if the g-code is fine but the printer refuses to print it, there's nothing we can really do about it. We'd need to know what is wrong about Cura's g-code to fix it.

closed 2022-01-17 by no-response[bot]