Ultimaker/Cura #11146: Skirt  Line Count (1 = 2 )

Sekisback, 2021-12-28:
### Application Version

4.13.b

### Platform

Linus

### Printer

Ender 3

### Reproduction steps

decreased the number of skirt lines to 1, will produce 2 skirt line
[image]
s

### Actual results

the skirt line will be 2

### Expected results

skirt line = 1

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[fussleiste_gerade_PLA_NZ-04_LH-02.zip](https://github.com/Ultimaker/Cura/files/7785691/fussleiste_gerade_PLA_NZ-04_LH-02.zip)

WhiteRabbit56, 2021-12-28:
If you enable the option "Skirt/Brim Minimum Length" and lower that from 250 mm does it still produce two skirt lines for you?

Sekisback, 2021-12-29:
you are right, the bug was in front of the screen

closed 2021-12-29 by Sekisback