Ultimaker/Cura #11147: Surface mode gaps (4.13 beta regression)

galvanized, 2021-12-28:
### Application Version

4.13.0 BETA

### Platform

Windows 10

### Printer

Creality Ender-3 Pro

### Reproduction steps

- Using Planeprint test print (["Wingtest PLA"](https://www.planeprint.com/print))
- Use default profile "Standard Quality - 0.2 mm" but with Surface Mode set to Surface

### Actual results

Intermittent missing layers on internal features when using surface mode on 1.13.0 BETA.
[image]
1.12.3 shown on left, 1.13.0 BETA shown on right

### Expected results

Behavior in 1.12.3 and earlier is as expected. Problem appears in 1.13.0 BETA.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project files: [Wingtest Comparison.zip](https://github.com/Ultimaker/Cura/files/7785979/Wingtest.Comparison.zip)

prometheanprints, 2022-01-02:
I am having same issues and even reverting to previous version of cuts causers other problems like connection over wifi to printer and sub extrusion on outer walls.

Vandresc, 2022-01-03:
Hi @galvanized thank you for your feedback, we will look into it.
Devs, see CURA-8860 for our internal reference.

Vandresc, 2022-01-03:
@prometheanprints , we are not aware of any connection issues, could you please create a ticket and fill in the bug template? Many thanks!

rburema, 2022-01-07:
Cherry-picked the fix commit for this to 4.13.0-final

Cause as determined and fixed by @casperlamboo was an extra call to simplification of open polylines introduced after 4.12.x

closed 2022-01-07 by rburema