Ultimaker/Cura #11149: Custom Profiles lost during beta upgrade

StuartP34, 2021-12-29:
### Application Version

4.13

### Platform

windows 10 64bit

### Printer

Anycubic Vyper

### Reproduction steps

Updated 4.9 beta to 4.13
Selected Keep Files prior to automatic uninstall of 4.12
After install custom material profiles were lost.

### Actual results

Custom Material profiles lost

### Expected results

Updated 4.9 beta to 4.13
Selected Keep Files prior to automatic uninstall of 4.12
After install custom material profiles were lost.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2022-01-03:
Hi @StuartP34 thank you for your report.
Just to double check the folder 4.12 is gone here: `C:\Users\<your username>\AppData\Roaming\cura`?

When I click No during the uninstaller in the popup Do you want to remove all your configuration files, it will show still the 4.12 folder with all my configurations.
Also good to know is that you don't have to uninstall previous versions of Cura.

Ghostkeeper, 2022-01-05:
A log file would help with this issue. Especially if it includes the part where 4.13 was first updated.

It should also be noted that when making the upgrade to 4.13, Cura doesn't know that you'd like your 4.9 profiles to be used instead of the 4.12 profiles. It'll take the profiles you had in the most recent version of Cura that you had before. So if you have both 4.9 and 4.12 installed and then upgrade to 4.13, Cura will take the configuration you have in 4.12. This means that if you kept using 4.9 after 4.12 was first launched and added profiles there, those profiles will not be migrated to 4.13 because they are not in the 4.12 configuration.

closed 2022-01-17 by no-response[bot]