Ultimaker/Cura #11150: Uneven Z seam and stuttering with simple model in Arachne beta

daniels220, 2021-12-29:
### Application Version

Arachne engine beta 2

### Platform

MacOS Big Sur

### Printer

Creality CR-10s

### Reproduction steps

Okay, so first off, searching around it looks like there is already an issue in your backlog for some of this, but, I think this submission contains (a) greater detail that may be helpful, and (b) some related issues not addressed elsewhere that I can see. Similar issues I'm aware of include #8010, #9555, #10181.

I sliced a very simple model—just a thin-walled tapered cylinder (a test piece for a vacuum-hose adapter)—with the Arachne engine beta, and found when I printed it that the printer was making a lot of extraneous travel moves between the two wall lines, and also stuttering in the middle of the extrusion moves.

### Actual results

Looking more closely at the preview, the first problem at least is obvious—the start locations of each line jitter significantly between layers and between the two concentric lines:

[image]

That layer is one of the most extreme, with the two concentric lines starting almost a centimeter apart, but all of them have _some_ separation. Frustratingly, nothing I do to the Z seam settings helps, and in fact selecting "Shortest" makes the problem _dramatically_ worse:

[image]

The worst part—and one I've not seen mentioned elsewhere—is that this applies to the skirt/brim lines as well: each successive line starts in a different place, resulting in the printer making sudden travel moves each time around rather than transitioning smoothly from one line to the next, which can easily knock the brim loose from the bed if adhesion is at all questionable. Not only that, but Arachne will sometimes print skirt/brim lines out of order—that is, numbering the lines with 1 being the innermost and counting upward, I've seen it print 1, 3, 5, 4, 2, leaving gaps and then filling them back in, rather than 1, 2, 3, 4, 5 or 5, 4, 3, 2, 1. This also dramatically increases the chances of knocking the brim loose before it can fully adhere. Changing the adhesion type on this very model to Brim produces that behavior, in fact:

[image]

### Expected results

Cura 4.12.1 handles the situation perfectly well:

[image]

This is with "Shortest" Z seam, and interestingly "User Specified" produces _some_ of the same jitter, depending on the chosen location—front-left produces the exact same result as shortest, and most of the other options are basically fine, but back-right is less than ideal, though still far better than what Arachne produces:

[image]

I've experienced having the "layer start position" influence this Z seam jitter in the past, as well—if that becomes relevant I can try to chase down the exact behavior further. Though it seems this is likely explainable by the comments on #8010...

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

All of this was with completely stock settings (I did a full reset of Cura by renaming ~/Library/Application Support/cura), except changes to Z seam settings as indicated.

Raw model: [TaperedHollowCylinder.3mf.zip](https://github.com/Ultimaker/Cura/files/7787419/TaperedHollowCylinder.3mf.zip)

Project as saved by Cura 4.12: 
[Cura412Project.3mf.zip](https://github.com/Ultimaker/Cura/files/7787441/Cura412Project.3mf.zip)

Project as saved by the Arachne beta: 
[ArachneProject.3mf.zip](https://github.com/Ultimaker/Cura/files/7787442/ArachneProject.3mf.zip)

Now, about the stuttering. I also noticed that the gcode output from the Arachne engine beta is a full 1.2MB larger than that from Cura 4.12. Certainly part of this is from the travel moves, but...aha! Looking at the code, I see that Arachne is causing the feed rate to constantly vary extremely slightly—almost certainly far too slightly for the stepper drivers to actually reproduce—which may be contributing to the problem:

```
;LAYER:1
M106 S85
;TYPE:WALL-OUTER
;MESH:178vac.3mf
G1 F1350 X166.238 Y166.272 E39.1923
G1 X165.996 Y166.517 E39.20661
G1 X165.138 Y167.301 E39.25493
G1 **F1352.7** X164.858 Y167.549 E39.27045
G1 X163.973 Y168.253 E39.31737
G1 X163.651 Y168.502 E39.33426
G1 X162.752 Y169.126 E39.37967
G1 X162.387 Y169.371 E39.39791
G1 **F1350** X161.476 Y169.918 E39.44209
G1 **F1352.7** X161.065 Y170.156 E39.46179
G1 X160.146 Y170.625 E39.5046
G1 **F1350** X159.692 Y170.849 E39.52565
G1 X158.775 Y171.245 E39.56718
G1 **F1352.7** X158.277 Y171.451 E39.58954
G1 **F1350** X157.367 Y171.774 E39.62969
...etc...
```

(Emphasis mine.) The Cura 4.12 output is similar (though the coordinates are different because the Z seam is in a different location), but the feed rate doesn't change once it is initially set.

But clearly that's not the whole of the issue, because turning flow rate compensation down to 0 gets rid of the feedrate jitter but the output is still 0.9MB larger. I can't so easily tell whether this is a result of higher-precision moves (dividing the circle into more segments), or if something else is going on, but doubling the "maximum resolution" (which, confusingly, reduces "resolution" as most people think of it, the help makes it very clear that it's the "minimum length of a segment after slicing", so longer is rougher) setting only makes the file 100kB smaller, still much bigger than 4.12's output.

There might be a clue here in the "flow" mode of preview display—Cura 4.12 is perfectly even, while Arachne is all patchy and weird, even with flow rate compensation turned all the way up (with it turned off I get hilarious visual artifacts, but that's a separate issue):

Cura 4.12:

[image]

Arachne beta:

[image]

Might as well include the gcode files, though I imagine you can reproduce them: [GCode.zip](https://github.com/Ultimaker/Cura/files/7787497/GCode.zip)

fvrmr, 2022-01-03:
Hi @daniels220 thank you for your detailed bug report. 
I can reproduce your issue with the seam. I will bring this up with the team.

About the brim see this issue: https://github.com/Ultimaker/Cura/discussions/8965#discussioncomment-237369 and this is indeed still in the second Beta. We have a ticket related to this on our backlog.

The visual lines are also a know issue see: https://github.com/Ultimaker/Cura/discussions/10931#discussioncomment-1748963

daniels220, 2022-01-04:
Great, thank you.

I realize now that the second issue I mention here, the flow/stuttering issues, should really be, well, a second issue, so I'll edit it out of here and submit it that way.

fvrmr, 2022-01-07:
I have created a ticket on our backlog for this.
Devs see CURA-8875

fvrmr, 2022-05-17:
The seam should be better in 5.0 now! So closing this issue. 
Here you can find the builds: https://github.com/Ultimaker/Cura/releases/tag/5.0.0

closed 2022-05-17 by fvrmr