Ultimaker/Cura #11152: Fan index not used in m106 when cooling fan number is 0

ladismrkolj, 2021-12-29:
### Application Version

4.12.1

### Platform

Mac/Win10

### Printer

Ender 5 - with skr pro

### Reproduction steps

When using two extruders with single part cooling fan, I set Cooling fan number to 0 for both extruders. 

### Actual results

M106 is sent without P0 index, therefore the fan only turns on for extruder T0, for T1 it doesn't work.

### Expected results

I expected that both extruders would turn on same fan when active.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It works as expected if cooling fan number is other than 0

GregValiant, 2021-12-29:
From the MarlinFW.org description of M106:
"Turn on one of the fans and set its speed. If no fan index is given, the print cooling fan is selected."
It would seem that the gcode is correct and that P0 isn't required to turn on fan 0 whether T0 or T1 is active, but that P1 is required to turn on fan 1 when T1 is active.  That's what I get in the gcode for my "virtual" Ender 3 Pro with dual extruders.
When configured with dual extruders and dual hot ends with T0 fan 0 and T1 fan 1 the gcode is:
T1
G92 E0
M106 S255 P1
-----and for T0
T0
G92 E0
M106 S255
Per the definition from Marlin that should work and activate each fan per the active extruder.
When set up as a 2-in-1-out system with shared heater and nozzle the M106 lines are in accordance with the cooling settings and ignore the T#'s.  They have no "P" parameter and so should activate the layer cooling fan.

I think more information is needed.
What firmware is the printer running?
Is your firmware set up with a single layer cooling fan?
Does the printer have true dual hot ends or is it a 2-in-1-out model with shared heater and nozzle?
Do the M106 lines look correct when T0 is disabled?

You have customized your printer.  It's a third party printer with a 4th party mainboard, probably 5th party firmware and was altered by a 6th party.  That being said - what I see in the gcode I generate (with my virtual machine) should work.

ladismrkolj, 2021-12-29:
Hi, my issue is similar to #8521. Let me clarify some things: I have Ender 5 with e3d chimera dual extruder (2 in 2 out). For part cooling both nozzles share one fan, which is a popular solution (e.g. https://www.thingiverse.com/thing:2947851), since it really isn't necessary to have independent fan on each nozzle. For my mainboard I have BTT SKR PRO, which has 3 pwm fan ports, correctly defined in Marlin as Fan0-2.

When having multiple fan pins defined in marlin, the M106/107 command defaults to `_MIN(active_extruder, FAN_COUNT - 1)`, as you can see, when T0 is active extruder, M106 without argument P effects Fan0, but when T1 is active extruder, M106 without argument P effect Fan1. 

For a month I bypassed this issue by only having fan0 defined in marlin, therefore m106/107 worked as expected, but now I need fan1 for heated chamber.

The P argument is passed to M106 in cura if cooling fan number is anything other than 0, therefore I don't see the problem why not pass it if cooling fan number is 0.

GregValiant, 2021-12-29:
That helped.
So your Fan1 is not a layer cooling fan but rather it's an enclosure exhaust fan and it is an actual physical fan that you have plugged into the F1 socket on the mainboard?  Out of curiosity - Is it thermostatically controlled?
Do you have "heated enclosure" enabled in the Cura Machine Settings?  That would add an M141 line to the gcode and Search and Replace could be used to find it and add an M106 P1 Sxxx right after the M141 line.  A second instance of Search and Replace could be used to turn it off at the end of the print.

ladismrkolj, 2021-12-29:
> So your Fan1 is not a layer cooling fan but rather it's an enclosure exhaust fan and it is an actual physical fan that you have plugged into the F1 socket on the mainboard?
Yes that is correct. It is thermostatically controlled and I have "heated enclosure" enabled in cura. However I am using a custom start g-code, so I added M141 and M191 where I think they fit best.

Search and Replace would most certainly work, however this is more a temporary workaround and at least for me not the satisfactory long-term solution.
I think the proper solution would be to be able to choose in cura whether you would like to force the use of `P0` or is that "0" in the Cooling fan number in a way just "disabled/default" option. Because right now Cooling fan number only really works if you choose non 0 index. 0 just means "do not choose fan index", which makes little sense as all fan indexes should be treated equally.

smartavionics, 2021-12-29:
See discussion around https://github.com/Ultimaker/Cura/issues/4259#issuecomment-426556019

ladismrkolj, 2021-12-29:
> See discussion around [#4259 (comment)](https://github.com/Ultimaker/Cura/issues/4259#issuecomment-426556019)

How about you change the label "Cooling fan number" to "Custom cooling fan number" and -1 is disabled/don't send P parameter, any other non-negative number would however force the use of P parameter?

Ghostkeeper, 2022-01-05:
As you say:

> When having multiple fan pins defined in marlin, the M106/107 command defaults to _MIN(active_extruder, FAN_COUNT - 1), as you can see, when T0 is active extruder, M106 without argument P effects Fan0, but when T1 is active extruder, M106 without argument P effect Fan1. But that is the problem, because Fan1 isn't part cooling fan on my setup...

And as [Marlin's documentation for M106](https://marlinfw.org/docs/gcode/M106.html) says:

> If no fan index is given, the print cooling fan is selected.

If it can select Fan1 when no fan index is given, and Fan1 is not a print cooling fan, then that is a direct conflict with the g-code specification. I think this problem is on your side then. It seems that Marlin expects that both Fan0 and Fan1 are print cooling fans.

> The P argument is passed to M106 in cura if cooling fan number is anything other than 0, therefore I don't see the problem why not pass it if cooling fan number is 0.

The problem with this is that this can give errors for printers that don't implement the P parameter. Many printers just stop the print when they encounter g-code that they don't understand. This prevents people from printing altogether with certain brands because someone somewhere connected up a fan wrongly in their custom printer mods.

Now we could build more settings that turn this behaviour on or off, but if it's just for one printer that has wrongly connected fans, I'd say that the workaround with Search and Replace should do fine.

GregValiant, 2022-01-05:
@ladismrkolj - If it was mine I might be inclined to wire the enclosure exhaust fan to the Fan 2 connection on the MB.  With M106 P2 S255 in the StartUp G-Code and M106 P2 S0 in the Ending G-Code it would be powered and shut off with the print.  That would eliminate the need for Search and Replace and Cura would never inadvertently refer to "P2".  Just a thought anyway.

ladismrkolj, 2022-01-05:
`If it can select Fan1 when no fan index is given, and Fan1 is not a print cooling fan, then that is a direct conflict with the g-code specification given by your firmware.` This is something I totally agree with. This is the root of the problem, however if every printer would have correctly defined or better "paired" each hotend to correct fan, there wouldn't be any need to define custom fan numbers in cura at all? Therefore, if someone once needed to have manual fan number defined for a hotend, this is the same as my problem, except that they needed to define a manual fan that is on other position than 0. The only way (AFAIK) to reduce the number of fans defined in cura without editing the board configuration file is through the `NUM_M106_FANS` but until chamber fan and cooler fan are coded diffrently in marlin, there is no workaround.