Ultimaker/Cura #11155: Modifier mesh should allow 0 perims setting

jaggzh, 2021-12-29:
### Is your feature request related to a problem?

While we can adjust perimeters count/thickness, we can't set it to 0.

### Describe the solution you'd like

Allowing a modifier mesh to set perims to 0 will allow us to "remove" surfaces of a model.
The goal of this is to allow the Lightning infill to be used for hollow shells, until Lightning supports are available.

[image]
Here's a mouse cover I have to print in parts (or it's an impossibly large amount of support structure, and quite a mess to clean up).

[image]
Here's part of it, with a test modifier mesh to remove the perimeters off part of it, so lightning supports can handle the top overhangs of the lilac-colored part.

[image]
Here you can see Cura considers the 0.0 settings invalid.
(The 50% infill was just to visualize the area to ensure it was being affected by the modifier mesh).

### Describe alternatives you've considered

I don't know an alternative to this -- not until we get Lightning supports.

### Affected users and/or printers

(This is not directly related to a specific printer.)

jaggzh, 2021-12-29:
The idea would be to have removed the perimeters of the object, so Lightning infill could support just the top:
[image]

fvrmr, 2022-01-04:
Hi @jaggzh thank you for another feature request.
When using the infill mesh only in modify settings for overlap it will only affect the infill. You can select cutting mesh and set wall thickness to 0 it will not print your wall where the modifier mesh overlaps with your model. 
For the lightning infill you can change the lighting infill support angle to a higher value to just support the top. 

Let me know if that could work for you.

Ghostkeeper, 2022-01-05:
> Here you can see Cura considers the 0.0 settings invalid.

This is not true. It colours it orange, indicating that it expects that it will not print well. But it does allow you to slice and print it.

And that seems to work correctly with your use case:
[image]

It does, of course, suffer a the problem I outlined [here](https://github.com/Ultimaker/Cura/issues/11151#issuecomment-1005592738). The lightning infill lines will hang in mid-air since they try to attach to the side of the infill area.

closed 2022-01-18 by no-response[bot]