Ultimaker/Cura #11157: Top Surface Skin Layer not working in arachne beta 2 99.10.0

TwomD, 2021-12-29:
### Application Version

arachne beta 2 99.10.0

### Platform

win 10

### Printer

ender 3 pro

### Reproduction steps

i cant get the top surface skin layer to work in the Arachne beta. it ignores all settings for the top surface skin layer and just uses the "normal" top layer settings.

if i save the project as a *.3mf and open it in cura v 4.12.0 it works and it uses the right settings ( 1 top surface skin layer, slower speed, different line with )

is this a known bug ? ( couldn't find anything about it )

### Actual results

arachne: 
[image]

### Expected results

cura 4.12.0:
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

the project file:
[CE3PRO_Lüftungsschlauch Durchführung _ teil1.zip](https://github.com/Ultimaker/Cura/files/7790641/CE3PRO_Luftungsschlauch.Durchfuhrung._.teil1.zip)

fvrmr, 2022-01-03:
Hi @TwomD thank you for your bug report. 
We discovered this as well and have ticket on our backlog related to this. 
I will link this to our ticket.
Devs see CURA-8839

TwomD, 2022-01-03:
Good to know :)

fvrmr, 2022-05-17:
This should be fixed in the 5.0 release: https://github.com/Ultimaker/Cura/releases/tag/5.0.0
So closing this issue!

closed 2022-05-17 by fvrmr