Ultimaker/Cura #11159: Overlap modifier ignoring settings, again

BloodyRain2k, 2021-12-30:
### Application Version

4.12.1

### Platform

Win 7 x64

### Printer

CR-6

### Reproduction steps

I was trying to correct some infill shenanigans that Cura was making, and this just showed multiple issues again with the overlap modifier.

[image]

The first problem was that I wanted to get these yellow filler lines in the upgoing shaft walls away, because they're structually not needed and Cura still generates them messy.
Because these only showed up when I had infill > 0, which I needed or the lower portion would be started mid-air, I attempted to overlap that section with a block that had Infill set to 0.

But of course the overlap modifier hates me and this didn't work. The fillers stayed regardless of whatever I've set the overlap modifier to.

So I tried the inverse and disable infill for the base and instead used the block to enable infill for the bottom section that needed it.
That worked, if filling the complete area with a solid block of infill. For whatever reason...

[image]

When the attempt above initially failed, I changed the mode to "Cutting Mesh" and that's when Cura decided to completely ignore the setting for "Top / Bottom Directions" from both, the base AND the overlap settings. See actual results.

### Actual results

[image]

Here the infill's properly done, even when I forgot to show it, but instead ALL except the first / last top / bottom layers are oriented wrong. And changing the directions doesn't change anything.

### Expected results

[image]

The reason THIS works now is because I've actually set the layers in the modifier.
The "Top / Bottom Thickness" setting alone doesn't do anything, and the seperate ones for just top and bottom aren't re-addable after you removed them with the minus.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Gap_Measurer_Body.zip](https://github.com/Ultimaker/Cura/files/7791118/Gap_Measurer_Body.zip)

fvrmr, 2022-01-04:
Hi @BloodyRain2k thank you for your report.

> The first problem was that I wanted to get these yellow filler lines in the upgoing shaft walls away, because they're structually not needed and Cura still generates them messy.

If you mean the lines in the top layer, it is indeed messy. You can use our Arachne build with variable line width where these lines will not be generated. 

I am a bit confused what is happening to your model with the modify mesh. Your project file is showing your expected results or not? 
If so could you share a project file where it is not working properly. Thank you in advance.

BloodyRain2k, 2022-01-04:
> Your project file is showing your expected results or not?

I wouldn't call bottom layers that ignore any and all direction input "expected".
And that's the state in which I saved the project, and the state I get in 4.12.1 when I load the uploaded file.

Ghostkeeper, 2022-01-05:
At first glance, with your project file, this looks like a duplicate of https://github.com/Ultimaker/Cura/issues/9219. One at a time mode doesn't work with modifier meshes unless the models are grouped. But I'll dig in.

> Because these only showed up when I had infill > 0, which I needed or the lower portion would be started mid-air, I attempted to overlap that section with a block that had Infill set to 0.
>
> But of [...]

I'm unable to reproduce this once the models are grouped. When I move your cutting mesh up, set the normal infill density to 10% and the cutting mesh's infill density to 0%, it does indeed cause the infill gap filling to disappear as you would expect:
[image]
[Project file (in a zip)](https://github.com/Ultimaker/Cura/files/7815568/infill_0_removes_infill_gap_filling.zip)

So this part is indeed duplicating #9219.

> That worked, if filling the complete area with a solid block of infill. For whatever reason...

It's cutting the mesh into pieces, resulting in an extra bottom side and an extra top side in the middle of the model. This is currently expected behaviour, since the cutting mesh literally cuts the model into pieces. There is a petition to change that behaviour here: https://github.com/Ultimaker/Cura/issues/9218

> When the attempt above initially failed, I changed the mode to "Cutting Mesh" and that's when Cura decided to completely ignore the setting for "Top / Bottom Directions" from both, the base AND the overlap settings. See actual results.

It's bridging. This is duplicating https://github.com/Ultimaker/Cura/issues/9083.

For the future, it would help if you looked if your issue was already reported elsewhere. This allows you to add to the conversation! For tracking it also helps to report only one bug per issue.

closed 2022-01-05 by Ghostkeeper