Ultimaker/Cura #11160: Critical Z heights

archer183, 2021-12-30:
### Is your feature request related to a problem?

Manually drawn support structures with narrow horizontal gaps are frequently ignored, resulting in no top/bottom layers where they should be.  variable Z height experimental setting does sometimes sort of work, but it tends to leave the top/bottom layers too thin at these locations.   

### Describe the solution you'd like

Allow the input of critical Z heights or the identification of user selected horizontal surfaces.   Z height will only be adapted in the region of these locations, with the Z height preferentially set as close to nominal as possible for the nearest M or N layers to the critical Z heights, where M or N are the number of bottom and top layers, respectively.  The net impact will be layers near these critical Z heights will be close to nominal and layers between critical heights will be adjusted to fit.  this will result in relatively small layer height differences generally while still hitting critical heights not divisible by the layer height value.

### Describe alternatives you've considered

Variable Z height results in unacceptably thin top/bottom layers and unnecessary refinement elsewhere in the model.   generally to hit my critical heights, the adaption settings needs to be set to a large range, resulting in a lot of very thin layers and inadequate top/bottom thickness in these regions.

### Affected users and/or printers

Users with critical Z heights not evenly divisible by layer height

archer183, 2022-01-06:
[example.zip](https://github.com/Ultimaker/Cura/files/7819383/example.zip)

in this slightly silly  example, if sliced at 0.2mm, most of the horizontal surfaces will not be at the correct location with a fixed layer height.  with adaptive layers, the layers will not distribute well due the curve that is noncritical.    ideal result would be even layer distribution (as close as possible) between each critical height   For example, on the 5.02mm tall wall segment, there should ideally be 26 layers at 0.19308mm thick or 24 layers at 0.209167 thick rather than 25 layers at 0.2.

see screen cap below where horizontal layers are not aligned with the horizontal surfaces due to the surface heights not being divisible by 0.2mm

[image]

adaptive layers really doesn't do what is needed, as seen below due partly due to the inability to tell it to ignore other curvy parts of the model and only hit the Z elevations that are critical to function.   Preview below also shows adaptive layers moving outside teh lower Z boundary on overhung surfaces (third image)
[image]

[image]

Ghostkeeper, 2022-01-11:
So as I understand, this issue is about dimensional accuracy in the Z direction, due to the rounding of Z coordinates to layer heights.

Currently, Adaptive Layers optimises the layer thicknesses with the goal to reach a target maximum horizontal shift per layer: Adaptive Layers Topography Size. For instance, if the Topography Size is set to 0.2mm, it will attempt to change the layer thicknesses such that every slope causes the horizontal distance between the edge of layers to be 0.2mm (as far as its limits allow).

The desired behaviour in this ticket is to change the layer thicknesses such that major horizontal planes end up at the correct height. There's 3 obstacles that will have to be tackled when implementing this:

* We can't change the layer thickness instantly, but have to transition slowly towards different layer thicknesses according to the Adaptive Layers Maximum Variation setting. This is already a problem for the current implementation too: https://github.com/Ultimaker/Cura/issues/6641 . However, if the optimisation goal is only about the accuracy of one layer, that optimisation is entirely defeated by the Adaptive Layers Maximum Variation. If this is to be fixed, the lookahead becomes more complex to start transitioning early enough to end up at an exact layer Z coordinate.
* This optimisation clashes with the current optimisation. What if a model has a steep slope that requires high layer height, but also has a horizontal surface that requires accurate Z positioning? Or the other way around: What if the model has a shallow slope that would require a lower layer thickness, but the correct Z position can only be achieved by increasing the layer thickness?
* You can have cases where models have multiple horizontal surfaces at almost - but not quite - the same height. Maybe one of them is achievable, but the other is not then. Which one should be optimised? The biggest surface, maybe? This can become quite a big problem since all models are triangulated in Cura, and most models have horizontal surfaces even for something tiny like the bottom of a cylinder or screw hole. If it is optimised, this would be a global optimisation problem, which can be both complex to implement and expensive to compute.

fvrmr, 2022-01-25:
We have discussed it with the Cura team and also with the team about print profiles and we agree this is something we would like to implement. There is a ticket on our backlog that is related to this.

fvrmr, 2022-02-01:
It is related to CURA-5104