Ultimaker/Cura #11161: (4.1.2) Dual extruder prime tower superfluous layers

mah115, 2021-12-30:
### Application Version

4.1.2

### Platform

Windows 10

### Printer

Weedo X40

### Reproduction steps

Enable dual extruder prime tower, slice.

### Actual results

[image]
[image]

If I swap the adhesion tendency order, now both of the materials get prime tower layers where they don't do anything (also, the prime tower bring uses the wrong extruder).
[image]

### Expected results

The prime tower should not extend past the dual material layers.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_Prime_Tower_Bug.zip](https://github.com/Ultimaker/Cura/files/7792293/CFFFP_Prime_Tower_Bug.zip)

fvrmr, 2022-01-03:
Hi @mah115 thank you for your report. 
The first time I opened your project file it was showing the prime tower correct for me. The prime tower stopped when there was not support printed. (Off course I forgot to make a screenshot )
The second time it showed the prime tower like your third screenshot. 

About the prime tower brim this is always the extruder which is set to Build Plate Adhesion Extruder, so in your case extruder 1. When changing this to extruder 2 it is showing correctly. 

I will bring this up for discussion, keep you posted!

fvrmr, 2022-01-07:
I have discussed this with the team and we decided to defer this issue, since it is unfortunately not a priority for us to fix since this is not happening with our printers. I hope you understand.

Ghostkeeper, 2022-01-11:
It also occurs with Ultimaker printers: [prime_tower_too_tall.zip](https://github.com/Ultimaker/Cura/files/7847356/prime_tower_too_tall.zip)

closed 2023-06-09 by github-actions[bot]