Ultimaker/Cura #11166: Loosing Default Profiles when switching Material Type makes no sense at all for me

awiltschek, 2021-12-30:
### Is your feature request related to a problem?

Yes i am loosing default profiles (as they are bound to the Material Type) when creating a new material (eg Carbon).
The default profiles of my printer should not get bound with the material unless i generated and saved own profiles for the specific material.
It also shows the material with orange background as invalid which makes again no sense.
So far i am useing profiles only with PLA although i print PETG/ASA/Carbon and so on.
So this binding of Material to profiles is either not implemented accordingly or nonsense.

### Describe the solution you'd like

I would like to have all relevant (0,12, 0,2, 0,28, 0,32) standard profiles for my printer bound to my printer.
When i create a new material i can bind the relevant standard profiles or specificly defined ones on my own in a seperate configuration dialog.
So i can manage and map profiles with material as i need it and not wrongly managed and removed by the application.

### Describe alternatives you've considered

not manage material and let them all be PLA (so this feature then is not used at all and therefore sensless) so all my special profiles and standard profiles are available although the material is not PLA.

### Affected users and/or printers

its a basic behaviour of cura so all users and printers are affected.

fvrmr, 2022-01-07:
Hi @awiltschek thank you for reaching out.
I assume you are creating your own custom materials. So the reason why this is happening is because Carbon is an unknown material type. And that's why it will show up in orange, that doesn't mean you can't use it. But we don't have any profiles for the material carbon, so Cura doesn't know what profiles to select, it will also say not support profiles. That doesn't mean you can't use this, but we(the Cura team or the community) didn't implement profiles for this material type. 

If you make a custom material and you fill in PETG for material type it will pick the already known profiles for PETG. 

Let me know if this makes sense to you!

awiltschek, 2022-01-07:
yes i understand your explanation, but the implementation doesnt make sense for me.
when a user creates a new material he should also have the possibility to select existing profiles he want wo adjust and config for this material.
in your implementation i am loosing (not showing me) all profiles (even the default profiles) and i need now to adjust ALL settings to build a new profile instead adjusting an existing one.
so only by fakeing (means using an existing material) and then using the still existing profiles i can make own materials and use them.
this doesnt make sense at all. so either you redesign the usage of material in combination with profiles or this featur „material“ is sensless at all.
i dont know if you understood my analysis about this feature.
so you combined printer with profile with material in a non optimal way.

Ghostkeeper, 2022-01-11:
By setting `"has_materials": true` in the printer's definition in the metadata, you're telling Cura that this printer has specialised profiles for each material. Some of these might have just 1 quality level it can be printed at, some might have 4 or 5. If there are 0 quality levels, Cura will say that that material is not supported by the current configuration (which is also a valid outcome).

Now it could be that your printer is lenient enough that this doesn't matter and it can print any material with the same (set of) profile(s). That's when `"has_materials": false` is useful.

It could also be that a specific material would work well enough with the profiles of a different material. This is where the Material Type comes into play: Two material profiles can be a different plastic, but fall in many of the same categories which means they can be printed with the same settings. In Cura you'd indicate that by the Material Type field. You could have two materials share the same "Semi-crystalline" material type and everything that is semi-crystalline will share the same set of profiles. Or you could have a plastic be PLA-like and share the material type "PLA" so that it gets the same set of profiles.

> a user can define materials and select existing profiles (including standard profiles for the printer) for further usage/tuning.

So indeed you would select the set of existing profiles by choosing the Material Type field.

I consider this a duplicate of https://github.com/Ultimaker/Cura/issues/8229, of which the current aimed solution is to turn the Material Type field into a drop-down (basically to select which set of profiles to use when this material is used) and to clarify its usage (yet undefined how, perhaps by changing the name of that field).

closed 2022-01-11 by Ghostkeeper