Ultimaker/Cura #11167: Cura 4.13.0 crashing when opening file

Apulo, 2021-12-30:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Custom - Voron Switchwire

### Reproduction steps

1. Click the open file icon.
2. Try to open a folder (not even a file)

### Actual results

Cura crashes to desktop

### Expected results

Cura opens the file/folder

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7794048/cura.log)

fieldOfView, 2021-12-30:
Perhaps this helps you:
https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

Apulo, 2021-12-30:
That did the trick, thanks!

closed 2021-12-31 by nallath