Ultimaker/Cura #11169: tool pathing gone

TroyIske, 2021-12-30:
### Application Version

4.12.1

### Platform

fedora linux

### Printer

creality ender 3

### Reproduction steps

1. I had 4.11 and learned of the feature to show layer height in the preview tab. this is awesome but now somethings wrong. the drop down menu for compatibility mode is no longer there. all i see is travels, helpers, shell, infill, etc. i want to see tool path and such.
2. my first attempt to fix was update to 4.12 No difference
3. i reached out to ultimaker and they had me turn on "force compatibility" This was already on so i turned off and reset. then turned on and reset. all resulting in no change.
4. the had me delete something and that too did nothing.
5. at this point they recommended i reach out to yall
5. i have now tried deleting everything with "cura" in the title 

attached are pics of no drop down menu

[image]
[image]

### Actual results

nothing happens

### Expected results

i want to be able to see tool pathing/more layer detail

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

thanks

fieldOfView, 2021-12-30:
The "Force compatibility mode" option needs to be turned **off**. Compatibility mode only has a very basic layer view.

TroyIske, 2021-12-30:
i just redownloaded and it is turned off. still no good

fvrmr, 2022-01-04:
Hi @TroyIske thank you for your report. 
If it is still showing your preview in compatibility mode, your driver or GPU does not support OpenGL 4.1, and this is required for the preview mode. 
This is something we won't fix, so I will close this issue. I hope you understand.

closed 2022-01-04 by fvrmr