Ultimaker/Cura #11170: Guidance needs to be updated.

rmfolsom, 2021-12-30:
### Application Version

4.12

### Platform

MacOS

### Printer

None

### Reproduction steps

Instructions say to unpack the printer definition files to a folder that does not exist. I am using a new MacBook Pro M1 running the latest OS. I created a folder per the guidance provided and unpacked the files there, but was doubtful from the outset since it appeared to a version specific path, i.e., 4.7. Using that did not work. Changed the path to 4.12. Still does not work.

### Actual results

Nothing

### Expected results

A Qidi printer definition would show up in the non-networked printers list.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

 If this is a dead end solution please take it down. If not please update it.

fvrmr, 2022-01-03:
Hi @rmfolsom thank you for your report. 
Which instructions are you following? Can you share a link?
You need to have these definition files in the following path: 
`Applications/<Cura version>/Contents/Resources/resources/definitions/`

rmfolsom, 2022-01-03:
Fenna,

I contacted Qidi. They provided the guidance I needed. The trick is that files are installed under the user, not the application. Command+Shift+G and then ~Library got me there.

The full path is: Macintosh HD/Users/user_name/Library/cura/4.Xcode

When adding the folders, make sure to use merge when folders with existing content are present.

There may be other ways, but this worked for me.

I was able to get the WiFi working using the instructions on GitHub. That is quite easy, however before I had installed Cura, I was using the Qidi version and when setting up that WiFi I used Test as the name. I then installed Cura and it used that name. Now it seems there is no way to change it something more meaningful. 

Cheers,
Randy

fvrmr, 2022-01-04:
Hi @rmfolsom thank you for your reply. 
We don't maintain the instructions from Qidi, so we can't update these :) 
I will close this issue since it is not related to our instructions. Hope you understand.

fvrmr, 2022-01-11:
Could you point me where the instructions are wrong.

rmfolsom, 2022-01-11:
Fenna,

It is the section below the plug-in instructions where guidance is given for installation of the files needed for Cura itself. Search for “4.7”. It was probably correct when originally published, but both MacOS and Cura have been updated. I am new to both Macs and 3D printing, so maybe I am missing something. Everything else is fine as far as I can tell. I have Cura and the plug-in working very well.

Cheers,
Randy

Ghostkeeper, 2022-01-11:
Cura has installation instructions, here: https://support.ultimaker.com/hc/en-us/articles/360011837399-Getting-started-with-Ultimaker-Cura

It doesn't provide any instructions about plug-ins (because a novice user doesn't need to know that plug-ins even exist). So I think you might be thinking about a different set of instructions? Perhaps you could provide a link of what you're looking at?

rmfolsom, 2022-01-12:
Perhaps you should keep your opinion about plug-ins and novices to yourself. There are some Cura plugins that every novice should install, for example Settings Guide. 

Cheers,
Randy

rmfolsom, 2022-01-12:
https://github.com/alkaes/QidiPrint

The section is as follows:

If you don't have Qidi printers available in Cura, unpack the Qidi Printer Definitions.zip to this specific folder:

Windows: %USERPROFILE%\AppData\Roaming\cura\4.7\
MacOS: ~/Library/Application Support/Cura/4.7/
Linux: /home/<username>/.local/share/cura/4.7/

The full path for MacOS is: Macintosh HD/Users/user_name/Library/cura/4.Xcode

What through me off was that I was not in the user directory, I was trying add it to the application directory. 

As mentioned earlier, Qidi will provide updated files. 

Cheers,
Randy

Ghostkeeper, 2022-01-13:
> There are some Cura plugins that every novice should install, for example Settings Guide.

Even then, they are expected to install that through the Marketplace, not by copy-pasting files into the configuration directory.

> https://github.com/alkaes/QidiPrint

So indeed, that is in QidiPrint's instruction, not in Cura's. We don't have control over what Alkaes posts in their QidiPrint plug-in's instructions. Please report your issue with that repository! https://github.com/alkaes/QidiPrint/issues
I don't think Alkaes will find it here.

I also think that the instructions are correct though, aside from the version number being outdated. But that is not for me to decide.

closed 2022-01-13 by Ghostkeeper