Ultimaker/Cura #11171: Bridge Second and Third Wall Speed

Calibr8tion, 2021-12-30:
### Is your feature request related to a problem?

On fast prints, the default wall speed may be too fast to remain on top of the first bridge wall.

### Describe the solution you'd like

Add the user adjustable option for second and third bridge wall speeds. 

### Describe alternatives you've considered

Must slow down the standard wall speed on the entire print. Not ideal for overall print times.

### Affected users and/or printers

Everyone

fvrmr, 2022-01-04:
Hi @Calibr8tion thank you for your feature request. 
I will bring this up for discussion. Keep you posted!

fvrmr, 2022-01-25:
I have discussed it with the Cura team and the team about print profiles and the bridging settings are planned for later this year. Most of the research being done right now is related to what happens after a bridge is finished and the skin begins again. We don't know yet if there is a need for it, so for now I will defer this request.

psiberfunk, 2022-04-08:
@fvrmr  Could I encourage you to take this up again ?  I ran into EXACTLY this problem today printing PETG.. to my surprise I couldnt slow down the wall speed, which is definitely what was killing my print.   I even went into GCODE and manually edited a few lines on a test print to see and slowing down the second and third (to a lesser degree) wall speeds on the bridge is absolutely key for printing PETG at 80 mm/sec base wall speeds.  I'd love if there's a way to tune this, and it doesn't seem like it should be too hard?

psiberfunk, 2022-04-08:
Here's a a perfect illustration of where these settings belong, too: 
[image]
Here, you can see that I get great bridging by setting the walls slow , and i'm able to set the skin slow on the second go too.. but what always ruins the bridge is that orange 80 mm/sec line on top of it, which is the second bridge layer wall which I can't tune for some bizarre reason.

psiberfunk, 2022-04-08:
Alternatively, for bridges like this, I would love if the parameters (fan/speed/flow) for the perimeters of the bridge could just be treated as if they were the same as the skin settings.

Chris-Jayden, 2022-10-02:
I am in agreement with both @Calibr8tion and @psiberfunk.
I've tried several hours and iterations of calibrating settings and using different bridging models, but the main thing that kills my bridges is how second and third layer bridges go way too fast - at my default printing speeds.

Please expedite and consider bringing this feature into upcoming versions of Cura, @fvrmr ~

psiberfunk, 2023-10-03:
This is still of value , and should be not too hard to implement.