Ultimaker/Cura #11172: Supports are hovering / not supported

Diggsey, 2021-12-30:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I have a model with a slanted base, and some structure above that needs supporting. All print settings are default for ender 3, except that I chose to "enable supports".

[image]

### Actual results

As you can see, the supports don't actually touch the ground, and so the printer just ejects filament into thin air:

[image]

### Expected results

The supports should touch the model or ground underneath them, so that they print correctly.

The model itself contains personal information so I'd rather not share...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional information.

GregValiant, 2021-12-31:
To the right of the "Settings Search" box is an icon with three lines on it.  Click on it and set the visibility to "All".
Go down to the "Support" section.  You will need to experiment to get what you want.  That area may require tree supports or you might get what you want by setting "Z overrides XY".  The XY distance and Z gap may come into play as well.  Enabling the Support Interface may help as well.

Diggsey, 2021-12-31:
@GregValiant if I set "Z overrides XY", then it gets closer, but it still hovers off the model. The "tree" support mode looks interesting, but I'm afraid that the trees are so unbalanced and unsupported that they will tip over during printing...

GregValiant, 2021-12-31:
Within the support settings did you "Enable Support Brim"?  I've had supports fall over but they were really skinny and I didn't actually expect them to survive.
If you use Normal supports then the setting for "Support Floor Distance" will be available.  You can set that to 0.  On large areas that can make the supports more difficult to remove but in your photo it looks like it is a limited area.

Diggsey, 2021-12-31:
> Within the support settings did you "Enable Support Brim"? I've had supports fall over but they were really skinny and I didn't actually expect them to survive.

The "Enable support brim" option seems to be enabled by default. The tree supports look like this:
[image]

So, they might be alright, it's just they go super far over before they join up with the model.

> If you use Normal supports then the setting for "Support Floor Distance" will be available. You can set that to 0. On large areas that can make the supports more difficult to remove but in your photo it looks like it is a limited area.

I couldn't see an option titled "Support Floor Distance", but I set the option "Support Z Distance -> Support Bottom Distance" to zero, and also set "Z overrides X/Y", but it still leaves a large gap:

[image]

Surely this is a bug in the software if it's producing un-printable Gcode?

fvrmr, 2022-01-04:
Hi @Diggsey thank you for your report. 
Could you share your project file (file - save project) here as a .zip here on Github? 
It is easier for me to reproduce and check your issue. Thank you in advance!

Diggsey, 2022-01-04:
@fvrmr I've removed the personal information from the model, and re-sliced it. It did not change the issue:

[CE3_blank_trophy.zip](https://github.com/Ultimaker/Cura/files/7808450/CE3_blank_trophy.zip)

GregValiant, 2022-01-04:
Minimum X Y distance 0.2, Z overrides XY, Roof interface 75% and floor interface at 50%, Support Overhang Angle 55°, Minimum Support Floor Area 2.0, there might have been a couple of others.
Here is a project file.  See what you make of it.  I think I liked it better with tree supports.  Changing the support angle to 55 got rid of the tall skinny supports for the butt.  I don't think they were going to hold up anyway.
[GV_blank_trophy.zip](https://github.com/Ultimaker/Cura/files/7809694/GV_blank_trophy.zip)

[image]

Ghostkeeper, 2022-01-11:
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/7191.

closed 2022-01-11 by Ghostkeeper