Ultimaker/Cura #11174: Skirt sits partly under model

DerbyOli, 2021-12-31:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Anycubic Mega S

### Reproduction steps

1. placed 3 objects on buildplate for slicing
2. sliced for printing

Remark: the project file I uploaded here is just one example. I had this behaviour with several projects during the past weeks, and I am still unsure how to reproduce this. This also happens with "Brim" selected instead "Skirt".

[skirtbug.zip](https://github.com/Ultimaker/Cura/files/7796146/skirtbug.zip)
[image]

### Actual results

Skirt positioning seems to ignore one of the three objects, so the skirt is positioned under this object (see attached screenshot also).

### Expected results

Skirt should be around all three objects, as usual.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

screenshot attached above (together with project file).

fieldOfView, 2021-12-31:
It looks like the effective vertices of those prongs do not touch the buildplate. You can fix this in the model, or by using the move tool to set the z to a negative value (eg -0.1) to cut a tiny bit from the bottom of the model.

gr5, 2021-12-31:
Okay so I just tried the included @DerbyOli project file (in the attached zip file) and there is no bug.  The part that the skirt goes "under" is not touching the build plate.  If they want the part to touch the build plate then the 4 portions that are over the skirt need to start at the same height as the rectangular portion that they are attached to.

Or you can set Z to a slightly negative value but to do that you have to first go into the cura preferences menu "preferences" "configure cura" select "general" and uncheck "automatically drop models to the build plate".

Someone should mark this as "not a bug" or "resolved".

DerbyOli, 2021-12-31:
thanks for looking into this. I was not aware that these parts were not touching the buildplate and thus will be ignored when calculating the skirt. I will set this to closed.

closed 2021-12-31 by DerbyOli