Ultimaker/Cura #11179: The coolingfan doesent start after selected layer

JohnnyEmilHansen, 2022-01-01:
### Application Version

4.13.0BETA

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

Cooling fan does not start on the selected layer. I set the fan to 100% after layer 4. but I have to manually start it on the printer.

### Actual results

I have tried and sliced ​​both with and without cooling but the result is exactly the same

### Expected results

Cooling should work the way I have sliced ​​it

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no

radry, 2022-01-01:
The problem can be two things:

Wrong Fan Number in Printer settings. Usually you have to set it to "0". (Machine Settings -> Extruder -> Cooling Fan Number)

Second Problem could be the (incorrect) behaviour of cura:

Instead of starting the fan at the specified height/layer, it gradually increses the fan speed starting from Layer 1 until it reached the specified layer.

JohnnyEmilHansen, 2022-01-01:
Ok, I think I found the problem. I had set the number of fans to 1 since it has a cooling fan, but I have now changed it to 0, Will test after the print that is on now is finished. thank you very much for the quick reply.

closed 2022-01-17 by no-response[bot]