Ultimaker/Cura #11180: Cooling Fan doesn't start at specified height/layer but increases gradually.

radry, 2022-01-01:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Enable Print cooling Setting
2. Set regular fan speed at anything you want above 1%
3. Set "Regular fan speed at layer" (or height) at anything above Layer 3

### Actual results

Fan "starts" (but not actually spins because it's too slow) at Layer 1 instead of the specified Layer.
Its speed increases gradually each Layer until the specified Layer is reached.

The problem is also when you set the regular fan speed very low like 30%, it will never start spinning because of the gradual increase. It would need a 100% "blip" to get it going. When I manually set the fan at 30% it works, but I assume because it gradually increases it doesn't.

### Expected results

Fan stays at 0% until the specified height/layer is reached, then starts with the specified "regular fan speed". Aditionally the fan must get an 100% impulse to get it spinning reliably for low speeds.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This bug has existed for a long time now. It doesn't make sense to have the fan spinning (actually it doesn't spin because it gradually increases and most fans need at least 30% to even start spinning) at heights below the specified one. 
Why would I set a certain height/layer for a fan to start when it ruins everything by spinning to early and potentially causing adhesion issues.

markecskace, 2022-01-05:
I agree. I do have the same  problem with part cooling fan. Gradual increase is not ok for like 30% speed. It is very difficult to set proper fan speed. I would like it to start at specified layer and not gradually. It is ok for PLA but not for PETG where slower fan speed is necessary.
SuperSlicer is going in the right direction with a bunch of settings.

fvrmr, 2022-01-07:
Hi @radry thank you for your report. 
It could be that this is set in your start g-code.

radry, 2022-01-08:
@fvrmr 
My Start Gcode is "G28", nothing else.

The gradual increase is added into the gcode by cura by default and can't be changed.

Ghostkeeper, 2022-01-12:
This starts to get into the realm of firmware too, though. If Cura's g-code output says `M105 S30`, then the firmware is supposed to make sure that the fan spins at 30% speed of its maximum. Usually that would involve [applying 100% speed for a fraction of a second](https://marlinfw.org/docs/configuration/configuration.html#pwm-fans-kickstart) to overcome the static friction. Depending on the fan, 30% duty cycle PWM may not be enough to make it spin, but all that's really necessary then is for the firmware to alternate e.g. 50% and 0% at a lower rate to cause it to spin the correct speed. The lower the duty cycle, the longer the pulse widths need to become. It's not really up to Cura to do that, I'd say.

Still, we could consider adding features in Cura to compensate for non-linear cooling rates vs. fan speed. If the fan spins at 30% rate, that doesn't necessarily mean the print cools down 30% as fast as it would at 100% fan speed. This sort of compensation is in Cura's scope. But then again, it's maybe not important enough to make it exactly linear there.

radry, 2022-01-12:
@Ghostkeeper 
Ok regarding the spike to 100% to make it spin you are right, that could be part of the firmware. 
The main problem is that cura gradually increases the fan speed every layer instead of turning on the fan at the specified layer.

> Maybe you should increase the Initial Fan Speed so that it doesn't start at 0%?

Why would I do that when I specifically want it to be 0% until Layer X and then start spinning with Y% ?

> You start at 30%, and increase to like 60% instead. Does that solve your problem?

No, the problem is, as I said, that cura incorrectly increases the fan speed every layer until the specified layer "regular fan speed at layer...".

Practical example: When printing ABS I want 0% fan speed for the first 1-2mm in height, above I want 30% (lowest possible my fan will spin at).  If I have any fan before that height, the print will warp. Thus 0% at the start until the specified height.

Ghostkeeper, 2022-01-14:
Ah, so the problem has nothing to do with PWM duty cycles not being linear to fan speed or with static friction.

The gradual increase is still necessary to prevent banding though. If the fan speed is increased too much too suddenly, you'll see a sudden change in the layers in the final print result where the fan kicked in. Making a gradual change is intended (so this is not a bug) and necessary to spread out the change.

This is the current model for the fan speeds over the course of a print (where the print gets smaller further up the print). The diagram is taken from the Settings Guide with slight modifications:
[image]

And this is what this feature request suggests:
[image]

It effectively adds another control point to the fan speed then. Doable, but it does make the fan control more complex for a user to understand as well.

radry, 2022-01-14:
@Ghostkeeper 
Just add an option (on/off) for gradual increase of fan speed.
on = current behaviour (default)
off = switch to regular fan speed at specified layer (no gradual increase in layers below)

closed 2022-01-25 by fvrmr

radry, 2022-01-25:
@fvrmr 
In that case the popup help or description in cura of the option "Regular fan speed at layer" should clearly state that on layers below that, the fan speed is gradually increased. The gradual increase of fan speed below the specified layer is not a Firmware issue but clearly caused by cura in the gcode. Currently it's not obvious to the user what is happening.

Ghostkeeper, 2022-02-01:
It currently says:

> At the layers below the fan speed gradually increases from Initial Fan Speed to Regular Fan Speed.

That seems clear to me.

MichaelHill525, 2022-05-10:
I agree that a hard off until a layer would be good. Printing in PC, I have parts that need absolute zero until I start getting some overhang.  The gradual increase is also understandable, but if I could have absolute zero until X layer and then start the gradual increase, it would be an immense help.

MichaelHill525, 2022-05-10:
radry, I have been going through my Gcode files in Bracket and just finding my M106 commands and deleting them until the desired layer.  At that point I start slowly increasing the fan over layers.  It has worked well for me so far. This works since I have files that I will print repeatedly, but I am sure it would be a pain if it was for every print you do.

mikebzh44, 2023-02-08:
Like @MichaelHill525 , I have to edit my GCode to move M106 line where I need it
@fvrmr : Is it so difficult to had an option like @radry posted above : option to switch from progressive fan to 'absolute value' fan ?