Ultimaker/Cura #11181: Performance issues Windows 10 / explorer - Significant Framerate drop while moving window / interacting with Cura

jbd1986, 2022-01-01:
### Application Version

4.12.1

### Platform

Windows 10 Pro: Version 10.0.19043 Build 19043  | System: i7 11700k @ 4.9ghz, 32GB RAM, RTX 3080TI, 144hz gsync monitor. 

### Printer

Anycubic Vyper - Doesn't matter though

### Reproduction steps

Description: 
When dragging other application windows around like explorer, chrome, office tools, etc, I get a silky smooth 144hz all the time (measured in GPU tools, and the monitor itself). When I move the Cura window, it's usually fine at 144fps, but if I move the window off screen at all while moving it, it will dip down to 10-60fps for a few seconds, before it ultimately returns to 144fps.  There is also a huge framedrop when leaving Cura alone (moving focus to some other app) and then coming back to Cura.

If I resize the Cura window larger (and don't go off-screen), it will resize smoothly at 144hz.  If I resize smaller, the framerate drop will occur.

### Actual results

Random FPS drops with Cura, even when no models are loaded, and no slicing has occurred.

### Expected results

When Cura is doing basically nothing, there should be no performance drop.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Debugging steps tried already: 
- Nvidia control panel, general settings set to prefer maximum performance
- Windows settings: enabled/disabled hardware acceleration, set power mode to performance, etc
- Verified that CPU/GPU usage are super low. And in fact, when the FPS drops, the usage actually gets lower, not higher.
- Lowering Mouse polling rate from 1000hz to 500, 250, 125. None of these made a difference.
- Disabled/enabled gsync and various related refresh rate settings
- Tried the latest Nvidia Game Ready driver, and then the latest studio driver. No difference.
- Updated Windows to newer versions, etc

jbd1986, 2022-01-01:
Here's a capture of the issue.  While some other 3D viewer apps have an initial lag, Cura is the only one that lags each time you move the window offscreen, resize it, or minimize/maximize to the system tray: https://www.youtube.com/watch?v=Th6eZVYDtn0

fvrmr, 2022-01-07:
Hi @jbd1986 thank you for your report.
Could you download the following plugin? https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging and when this plugin is installed do the same thing you did on your video and then share your log files?
Maybe this can help us.

jbd1986, 2022-01-07:
> Hi @jbd1986 thank you for your report. Could you download the following plugin? https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging and when this plugin is [...]

Hello @fvrmr, I've enabled extensive logging.  I'm not familiar with the logging system in Cura though.  Let me know exactly what you need.

jbd1986, 2022-01-07:
[cura.log](https://github.com/Ultimaker/Cura/files/7829793/cura.log)

I've found this log.  I added a comment before I began manipulating the window.  Didn't see anything relevant pop up in there.
========= Start moving/resizing HERE ===========

jbd1986, 2022-01-07:
Note, this issue is observable on my Work PC (another windows 10 PC, using integrated intel graphics on an intel i7 8700), in addition to the PC I listed in the original description.  Cura is MUCH slower on my work PC when resizing and moving the main window off-screen.  It basically becomes a slideshow.  And again, this is with no model loaded.

jbd1986, 2022-01-08:
@fvrmr FYI, disabling gsync on my gaming PC drastically improves (but does not solve) the framerate drop.  resizing down still has minor hiccups, and going off screen has major hiccups (but not quite as bad).

My work PC doesn't have gsync, so nothing to test there.

Ghostkeeper, 2022-01-12:
> Hello @fvrmr, I've enabled extensive logging. I'm not familiar with the logging system in Cura though. Let me know exactly what you need.

When that plug-in is installed, under the Extensions menu there will be a new menu item "Remote Support". You'll want to:
* Start the CPU profiler.
* Do whatever is necessary to trigger the slow-down. In your case, apparently, moving the window to a different monitor or resizing it.
* Stop the CPU profiler.
* Save the remote support archive (a .zip file) somewhere.
* Post that .zip file here.

This .zip file contains your log, but your log says nothing that is of use for frame rate. However it also contains a profiling of which functions were called in Cura's code and how long they took to execute. This may allow us to find out what it's working on.

> It really seems like the issue is tied to how Cura interacts with Windows explorer.

So this is the thing: Cura doesn't interact with Windows Explorer. Cura interacts with [Qt](https://www.qt.io/) and Qt interacts with Windows Explorer for us. So what we're looking at may just be a bug in Qt. If this is the case, we're probably not going to be able to fix it, but will be at the mercy of Qt's developers.

Another possibility is that Cura's rendering is simply slower than other applications. I know that resizing the window triggers a complete re-render for sure, both of the 3D scene and the interface. The 3D scene will be mostly up to your GPU but the interface involves a lot of CPU work too, so if that is the constraint you won't see it in the GPU workload. It's hard to see the drop of framerate in the YouTube video, by the way. We work at different refresh frequencies.

closed 2022-01-26 by no-response[bot]