Ultimaker/Cura #11182: Save windows size is not possible resize.

soundserie, 2022-01-01:
### Application Version

4.12.1

### Platform

MacOS 12.1

### Printer

Creality CR10 v2

### Reproduction steps

[image]

G-code save windows is not possible resize and file name is not possible modificative (is not show).
Version 4.12 have same issue.

### Actual results

File is save with for defined name (correctly save).

### Expected results

Windows size is possible change or default is windows size is bigger (vertical size).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Not file depended issus.

fvrmr, 2022-01-04:
Hi @soundserie thank you for your report.
I tried it myself but my save window is showing correctly and I can also resize my window. 
Are you maybe using an external monitor?

soundserie, 2022-01-04:
Hi, yes my setup have two monitors. But I have same trouble with one
monitor too (macbook pro with Intel cpu and egpu Ati).

BR Michal

Ghostkeeper, 2022-01-12:
This could be a bug in Qt then. We simply create the default QFileDialog:

https://github.com/Ultimaker/Uranium/blob/66da853cd9a04edd3a8a03526fac81e83c03f5aa/plugins/LocalFileOutputDevice/LocalFileOutputDevice.py#L56-L66

Maybe there are workarounds possible? We haven't seen this in our own testing though, so we don't know how we can reproduce this issue.

soundserie, 2022-01-12:
Hi, I tray today new version 4.13. Windows problem is not more present
(eGPU, Cura running on extern monitor).

Im very happy :-) THX
BR Michal