Ultimaker/Cura #11183: Option to select other languages or at least option to set in config file

VladimirVecera, 2022-01-01:
### Application Version

4.12

### Platform

Windows 10 64bit

### Printer

Creality Ender 3 v2

### Reproduction steps

The language setting does not contain everything from the ./i18n/ languages folder 

### Actual results

Nothing, I can't choose the other languages I would like to choose. Although the language is probably a beta version, I would like to be able to choose it. Which, of course, is not the case. 

### Expected results

Possibility to select other languages or at least the possibility of setting in the config file. Just have a choice! 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

How do I set up other languages that are in the i18n folder and can't be selected directly in the program itself?

[image]

fvrmr, 2022-01-07:
Hi @VladimirVecera thank you for report.
I will bring this up with the team. Keep you posted!

Ghostkeeper, 2022-01-12:
The Cura interface explicitly hides some languages because they are not complete. That is not up to our quality standards so languages that are not complete will not get shown.

You can still change the language in the `cura.cfg` file to languages that are not complete. In `C:\Users\<username>\AppData\Roaming\cura\<version>\cura.cfg`, you would add a field called `language` under `[general]` that refers to the language code, like so:

[image]

This allows you to select the languages that are incomplete (currently `fi_FI` and `hu_HU` and `pl_PL`).

closed 2022-01-12 by Ghostkeeper

fvrmr, 2022-01-25:
We have discussed it with the team and all the languages are selectable in the next release even if they are not completely translated. See the linked pull request above.