Ultimaker/Cura #11184: Pause at height not processing or generating gcode when Raft is selected and sliced

kevin-aus72, 2022-01-02:
### Application Version

4.13.0-beta

### Platform

windows 10

### Printer

cr-10s

### Reproduction steps

I have tried useing the Pause at height script 

With raft turned on 
I have been trying Pause at height  By layer 200 as a example

### Actual results

slice the model  and save no pause at height gcode generated

### Expected results

When i tun raft off the pause at height by layer 200 as a example works and generates pause at height gecode  

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7798761/cura.log)
[Component.zip](https://github.com/Ultimaker/Cura/files/7798773/Component.zip)

GregValiant, 2022-01-02:
When using "Raft" and if the part is located > 1 layer height above the build plate - there is no "LAYER:0" inserted into the gcode file.  In this project the part is located 17mm above the build plate.  (@kevin-aus72, why is that?)
When PauseAtHeight runs then "layers_started" stays false:

layers_started = False
....
# Fist positive layer reached
                if ";LAYER:0" in line:
                    layers_started = True

If the model rests on the build plate or is <= layer_height above the build plate and is on a raft then Layer:0 is inserted and it's all good.
When Skirt is enabled then LAYER:0 does get inserted into the gcode regardless of model placement and so "layers_started" becomes true and the Pause code is inserted.

kevin-aus72, 2022-01-02:
The project included is a model of a handle which a bolt is inserted at a height of about say 200 layers when pause at height is run.. The support comes up and allows the support blocker to block supports around the bolt hole and to support the handle.. Going on math and bolt length handle has to be 17mm of the plate .The raft is only used for support for the size of the print

GregValiant, 2022-01-02:
Thank you.  I was wondering what you were doing there.
I think the team is going to want to look at this.  In the mean time and as a work-around - with the Brim enabled and all the support being printed I think you are OK as far as stability while printing goes.  The raft would appear to be over-kill and it does look like the problem is in the raft generation.

kevin-aus72, 2022-01-03:
Thank you GregValiant . Yes brim i have extended the brim and it this should work.. A few more changes to the model and  will again. 
  
Why is does the raft start at layer -6 and then model start at layer 1
Why not start say what is on the bed and everything above the bed start at layer 1 because the layer height in the preview window includes the raft and starts at the build plate. manual inserting the pause at height commands layer numbers dont match ....

Sorry to ask these questions I know above my pay grade

fvrmr, 2022-01-07:
Hi @kevin-aus72 thank you for your report.
I will bring this up for discussion with the team. Keep you posted!

kevin-aus72, 2022-01-07:
@fvrmr  thank you .. 
I know it is a very small problem in software.. 
I am trying to to create a script to reorder the layer numbers and then call the pause at height script..
Will keep you posted

GregValiant, 2022-01-08:
I was playing with Search and Replace and got it to work (I think).  It needs to be first on the list of post-processors.
Search = ;LAYER:-1
Replace = ;LAYER:0\n;LAYER:-1
When PauseAtHeight runs it sees the "Layer:0" and so inserts the pause...at the wrong layer.
Now I think I know where your question "_Why is does the raft start at layer -6 and then model start at layer 1_" comes from.
This is line 340 and 341 from PauseAtHeight.py
if current_layer < pause_layer - nbr_negative_layers:
         continue
In PauseAtHeight - if I enter 50 as the layer to pause at then the pause is inserted at layer 44 as there are 6 negative layers in the gcode.  That is at best not intuitive at all.
I believe there are two things going on here.

1. The first layer of the model should be LAYER:0 as it is with when Build Plate Adhesion is set to anything other than Raft.
2. If I enter a layer number to pause the print then the print should pause at that layer as it does when anything but Raft is chosen for build plate adhesion.  (PS:  I have a bet with myself as to how that happened.)

kevin-aus72, 2022-01-08:
I dabble at some things and master of none

For the last couple of days i had been thinking at the data block passed in to the python script was a list data format
I was wondering why i was getting errors. Last night i just realized not LIST DATA  but it is a STRING DATA OBJECT   

This is not code this is my understanding of gcode postprocessing

This is my attempt at Trying to understand the start of the script fundamentals
For the people trying to make a script below code this is just a part of the data String object
It gets broken down in to 

layers using **layer.split("\n")** and then 

for loop to run through the string finding layer sections  **;LAYER:**

From here we can we edit gcode and then reassemble the data string object when we finish the script

The cura then processes the data string object and saves the gcode string to a file once the python script is closed

If you are trying to debug the script you need to look at the cura log files that is were the problems will be caught

  
**Data String Object**
[';FLAVOR:Marlin\n;TIME:13360\n;Filament used: 11.6059m\n;Layer height: 0.2\n;MINX:126.371\n;MINY:126.333\n;MINZ:0.36\n;MAXX:173.594\n;MAXY:173.553\n;MAXZ:89.91\n', ';Generated with Cura_SteamEngine 4.13.0-BETA\nM140 S60\nM105\nM190 S60\nM104 S200\nM105\nM109 S200\nM82 ;absolute extrusion mode\nM201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration\nM203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate\nM204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration\nM205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk\nM220 S100 ;Reset Feedrate\nM221 S100 ;Reset Flowrate\n\nG28 ;Home\n\nG92 E0 ;Reset Extruder\nG1 Z2.0 F3000 ;Move Z Axis up\nG1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position\nG1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line\nG1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little\nG1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line\nG92 E0 ;Reset Extruder\nG1 Z2.0 F3000 ;Move Z Axis up\n\nG92 E0\nG92 E0\nG1 F1500 E-6.5\n;LAYER_COUNT:443\n'] only part off

Ghostkeeper, 2022-01-12:
> Why is does the raft start at layer -6 and then model start at layer 1
Why not start say what is on the bed and everything above the bed start at layer 1 because the layer height in the preview window includes the raft and starts at the build plate. manual inserting the pause at height commands layer numbers dont match ....

History, basically. It was thought to be easier when there's a raft to just paste layers at the bottom, instead of shifting all of the layers upwards. It's also difficult to change because g-code consumers depend on Layer 0 being part of the model.

GregValiant, 2022-01-12:
"...consumers depend on Layer 0 being part of the model."
Having LAYER:0 as the first layer of the actual model is helpful and has always been consistent.  I know floating a model > 1 layer height above the build plate isn't a normal situation.  I found the lack of a Layer 0 in the gcode file surprising.  PauseAtHeight would seem to depend on having a Layer 0 in order to initialize itself, or know where to start.

kevin-aus72, 2022-01-13:
Thank you all for you Feed back it has been a little challenge for me python hmm python easy and head banging the table at the same time..

I am not a professional programmer use at your own choice
This is my attempt to correct my problem

Searches all data[ ] list Line by line and renumbers all layers starting at zero
Then changes the total layers count to the total of renumbered layers 
then moves back to a data list 

Must be first on the post processor list 
1)Layer_From_Zero_Renumber 
2)Pause at height

[Layers_From_Zero_Renumber.zip](https://github.com/Ultimaker/Cura/files/7862753/Layers_From_Zero_Renumber.zip)

fvrmr, 2022-01-14:
I have discussed this with the team and checked previous versions of Cura and it happens there as well. We didn't think this would be high priority, so we would defer this issue.
So it is good to see that you created your own script @kevin-aus72, well done!

closed 2023-06-09 by github-actions[bot]