Ultimaker/Cura #11186: cura not opening once downloaded

nhamilton1995, 2022-01-02:
### Application Version

4.12.1

### Platform

windows 11

### Printer

ender 3 pro 

### Reproduction steps

downloaded the new version of cura 4.12.1
uninstalled old version
installed new version 4.12.1
once the install was done cura auto opened everything worked fine 
shut my computer down for the night 
next day tried to open cura and it told me i have to download it again 
did the uninstall and reinstall works fine 

every time I close cura and try it reopen it, it makes me uninstall and reinstall  

### Actual results

nothing it make me install every time 

### Expected results

i would like it to be fixed so i can click the shortcut and open cura 
not have to uninstall and reinstall every time 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

please fix asap

fvrmr, 2022-01-04:
Hi @nhamilton1995 thank you for your report.
Could you check if this is happening when you are opening Cura.exe in the following folder: `C:\Program Files\Ultimaker Cura 4.12.1`
It looks like you are opening the installer instead of Cura.exe

Ghostkeeper, 2022-01-12:
Otherwise, a log file would be very useful too!

closed 2022-01-18 by no-response[bot]