Ultimaker/Cura #11187: Add a feature to add your own print bed model

Silithas, 2022-01-02:
### Is your feature request related to a problem?

It can get quite hard to see which orientation the print bed is in cura, and i want to request a feature to add your own model and texture for 3d print bed that of course turns transparent when looking underneath the model.

### Describe the solution you'd like

Being able to upload a model and textures for the model of the preview bed like prusaslicer can.

### Describe alternatives you've considered

Prusaslicer, but i prefer cura.

### Affected users and/or printers

Everyone and every printer technically.

fvrmr, 2022-01-07:
Hi @Silithas thank you for your request.
Do you want this for custom printers? Because most printers already have a mesh when they contribute their printer profiles to us.

Silithas, 2022-01-09:
> 

Custom as well as creality. Say i want it to have the same texture as the stock ender 3 pro bed, or a different fancy texture on it.

https://i.imgur.com/nIRaeFH.png Cause this looks too plain to me and at times, the dragon logo and ender text can look so translucent with a print on that i'd have to look underneath to tell which orientation i'm at.

Cause i'm using ender 3 pro, but custom due to new motherboard and custom marlin branch, so not really ender anymore apart from stock bed and stepper motors.

Ghostkeeper, 2022-01-12:
You can change the mesh in Cura fine. You just have to place a file in the correct spot. In Cura, you can go to Help -> Show configuration folder. There's a folder called `meshes` there and if you place a mesh called `creality_ender3.3mf` there I'd expect it'll get shown for the mesh. Or you could change the [definition](https://github.com/Ultimaker/Cura/blob/a1622c77ea7259ecb956acd6de07b7d34b7ac52b/resources/definitions/creality_ender3.def.json#L8) to point to a different mesh.

We consider making custom printer definitions to be expert enough that anyone that customises their printer can also edit JSON files. For this reason there is no graphical interface for it. Documentation on how to do this is [here](https://github.com/Ultimaker/Cura/wiki/Definition-Files-Explained).

closed 2022-01-12 by Ghostkeeper