Ultimaker/Cura #11188: Window Scale Bug

zaag1, 2022-01-02:
### Application Version

4.12.1

### Platform

MacOS 12.1

### Printer

Ender 3 v2

### Reproduction steps

1. Starting Cura after OS Reboot
2. Scale setting of print setting window is to large for native Macbook Display without external display.

### Actual results

Sometimes not possible to change the size of the printsetting menu. 

### Expected results

should be possible to change the window size with the mouse

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7799625/cura.log)

fvrmr, 2022-01-07:
Hi @zaag1 thank you for your report. 
I tried it on a Macbook with MacOS 12.1 and I couldn't reproduce this. 
Could you share a screenshot and does it not happen with an external display?

zaag1, 2022-01-15:
Hi, 
it happens with internal an external display. Sometime it is not possible to move an object in cura to another coordinate. I determined that I can fix the problem for the moment, if I move the MacOS window and go back to „fill out the screen“ (not fullscreen).

Ghostkeeper, 2022-01-17:
The scale of Cura's interface is dependent on the font size of your system. The minimum window size of Cura's main window is at the moment [80x48em](https://github.com/Ultimaker/Cura/blob/0501b71e13dbe1ae94dd4a24cb62c92bda4e7475/resources/themes/cura-light/theme.json#L481). An [em](https://en.wikipedia.org/wiki/Em_(typography)) is measured by asking Qt what the height is of the ascender of the system's font.

So it could indeed be that your computer's font size, DPI and screen size combine to making it too small for Cura to display its interface.

zaag1, 2022-01-17:
> The scale of Cura's interface is dependent on the font size of your system. The minimum window size of Cura's main window is at the moment 80x48em. An em is measured by asking Qt what the height is of the ascender of the system's font.

Other way around. No issue with window is smaller than screen. Issue if window is on whole screen (or MacOS fullscreen (green button)). 

Gruß

Andreas Maier

zaag1, 2022-01-17:
btw: Macbook Pro 13“ 2015 

Gruß

Andreas Maier

Ghostkeeper, 2022-01-21:
So if you make the window bigger, the setting list suddenly doesn't fit any more?

What happens if you gradually make the window bigger by dragging the corners of the window, instead of the MacOS maximising?

closed 2022-02-07 by no-response[bot]