Ultimaker/Cura #11190: Support interface has gaps if z-distance is set to 0

Doccos, 2022-01-03:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Selfmade

### Reproduction steps

Set Support interface extruder to EXT2
Set Support z Distance to 0 (i use soluble material)
Slice

### Actual results

When z-distance is 0 then there are gaps in the support interface.
If z-distance > 0 there are no gaps

### Expected results

a clean support interface when z-distance is set to 0

With z-distance 0.1
[image]
[image]

With z-distance 0

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file:
[blende.zip](https://github.com/Ultimaker/Cura/files/7800584/blende.zip)

fvrmr, 2022-01-07:
Hi @Doccos thank you for your report.
Could you share your printer definition file with me? I can't open your project file because it is missing your printer definition. 
And if I follow your reproduction steps I am not able to reproduce this.

You can find your printer definition here: `C:\Program Files\Ultimaker Cura 4.12.1\resources\definitions` and the definition files for your extruders here: `C:\Program Files\Ultimaker Cura 4.12.1\resources\extruders`

Doccos, 2022-01-07:
Thanks for the answer.

I tried it now with a clean installation of cura on a other PC.
As printer a choosed "Raise3D N2 Plus Dual"

I attach the new screenshot and project file here
[blende2.zip](https://github.com/Ultimaker/Cura/files/7829394/blende2.zip)
[image]

fvrmr, 2022-01-11:
Thank you, I can now reproduce your issue. 
I will bring it up for discussion. Keep you posted!

fvrmr, 2022-01-14:
I have discussed it with the team and decided to defer this issue. Since we believe it is an edge case and model specific. Also we think that it would affect the print quality, since the gaps are small.
I hope you understand.

sebastiankuzminsky, 2022-04-12:
Hi there, this bug is affecting us as well.
I can reproduce this by slicing in 4.13.1 for an Ultimaker S5, using the "Default / Fast" profile but with "Support Z Distance" set to 0.  (I also disabled Build Plate Adhesion and changed the Support Density to 100% to make the problem clearer, it happens without these changes but is harder to see.)
My model looks like this:
[image]
The top layer of the support has the most obvious problem, though there are glitches on lower layers as well.  The problem occurs whether or not Support Roofs are enabled:
[image]
This view from below shows the top layer of the support and the layer immediately above, showing clearly how parts of the model are unsupported:
[image]

[UMS5_35_10113_40_r24 (mm).3mf.zip](https://github.com/Ultimaker/Cura/files/8474574/UMS5_35_10113_40_r24.mm.3mf.zip)

sebastiankuzminsky, 2022-04-13:
Setting `Support Stair Step Height` to 0 works around the problem for me (and note that in my case, the faulty support is resting on the print bed, not on the model, so support stairs are not relevant).

closed 2023-06-09 by github-actions[bot]