Ultimaker/Cura #11191: Cura Slicing Error

ctucker1, 2022-01-03:
### Application Version

I'm not sure. On the loading screen it says "master"

### Platform

Linux (Raspberry Pi 4b)

### Printer

Ender 5

### Reproduction steps

I designed a new spool holder in Tinkercad and imported it to Cura (as an .stl).

### Actual results

After slicing it (with my normal settings), I got an error saying "Slicing failed with an unexpected error...". 

### Expected results

Instead of getting error, I expected an estimated print time and a button that allows me to save to a disk or removable device.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Here is the project as a .zip file:
[CE5_Spool Holder.zip](https://github.com/Ultimaker/Cura/files/7800652/CE5_Spool.Holder.zip)

Here is a screenshot of the error:
[image]

If you need a .stl of the file I was going to print, I can send it via email at goldenretrieverdude29@gmail.com since GitHub doesn't accept them.

GregValiant, 2022-01-04:
The "Master" version is a branch of Cura maintained by @smartavionics.  He may see your report here.  He has his own github site [HERE](https://github.com/smartavionics/Cura/issues).

smartavionics, 2022-01-04:
Hi, yes, it's my Cura build. Perhaps you are using an old build and a more recent build would be OK. I have just sliced your project on a Pi 4 and it completed OK.

[image]

If you continue to have problems, please open an issue on my repo and close this issue so to avoid spamming the UM folks.

smartavionics, 2022-01-04:
BTW, I see the project is using the Schwarz P infill pattern. That's a bad choice for this print, gyroid would yield a much stronger part and is quicker to print.

fvrmr, 2022-01-04:
Closing this issue since it is not related to our version!

closed 2022-01-04 by fvrmr

ctucker1, 2022-01-04:
I’m sorry for the trouble