Ultimaker/Cura #11192: menu déroulant du "mode détour"

Lolashz, 2022-01-03:
### Application Version

4.13.0 fr

### Platform

windows 10

### Printer

flsun SR

### Reproduction steps

[image]

### Actual results

x

### Expected results

x

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

x

fvrmr, 2022-01-06:
Hi @Lolashz thank you for your report.
You are using the 4.13 beta which is not translated yet. So it should be in the 4.13 stable. 
I will close this issue but if this is not fixed in 4.13  stable let me know.

closed 2022-01-06 by fvrmr

Lolashz, 2022-01-09:
hello,

i join you some screenshot 
capture--> the one i'm openning 4.13.0 beta
capture1--> the openning/loading with the version
capture2--> the software speaks to me in french and english :) , it thinks that i'm on an other version of cura
capture3--> the issue i've opened on github....

as you can see on the first screenshot, i've got older versions of cura, maybe it makes this issue ?

if you want me to run some test or other things let me know
i'm on windows 10

sorry for my bad english, i'm a french noob speaker ;)

Ghostkeeper, 2022-01-13:
> i join you some screenshot 

You can't share screenshots via e-mail replies to Github. To upload files such as screenshots, you'd have to go to the Github website: https://github.com/Ultimaker/Cura/issues/11192

Again, this is known. We translate Cura during the beta period, so new features won't be translated yet. They are translated in the stable release though.