Ultimaker/Cura #11197: AnyCubic i3 Mega S/Pro - Incorrect Profile for Printer

Empire-of-E-Projects, 2022-01-04:
### Application Version

Latest Version

### Platform

Windows 10

### Printer

AnyCubic i3 Mega S/Pro

### Reproduction steps

This issue is that the printer profile has incorrect maximum values for retraction and for initial printing temperatures.
 
• The user manual for the *Mega S/Pro, recommended 60mm/s to 100mm/s for regular printing.
 
• The community provided default profile for this printer has had maximum values set for the retraction speed to 40mms.
 
- The issue is that CURA will not let me set the recommendation from the user manual at 60mms.

It's being hard coded to lock at 40mms maximum, a full 20mms slower than the user manual recommended for a base mark.
 
It's impossible to set the correct speed for the printer retractions - after digging into the source code, i managed to fix this issue by manually editing the default profile for *Mega S/Pro
 
I Removed the 40mms maximum limit, and changed it to a more usable 100mms max speed
 
The printer manual suggested 90mms for ABS plastic - so I've updated my personal edition to include a maximum of 100mms.
 
I also found some strange calculation on the initial printing temps - the profile is set to print the first layer a full 10° hotter than the next layer.
 
They literally wrote "Temperature + 10" 
 
When it should be "Temperature - 5" 

### Actual results

Corrected the incorrect maximum values for AnyCubic i3 Mega S/Pro profile

### Expected results

The print quality will drastically increase and the correct settings can be chosen by the user.
 
The user can actually follow the correct recommendation by the manufacturer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

None.

Empire-of-E-Projects, 2022-01-04:
Here is my refined version of the profile with corrections.
> File should replace C:\Program Files\Ultimaker Cura 4.12.1\resources\definitions\anycubic_i3_mega_s.def
   

```
{
    "version": 2,
    "name": "Anycubic i3 Mega S/Pro",
    "inherits": "fdmprinter",
    "metadata":
    {
        "visible": true,
        "author": "Anycubic Enthusiast",
        "manufacturer": "Anycubic",
        "file_formats": "text/x-gcode",
        "platform": "anycubic_i3_mega_s_platform.3mf",
        "has_materials": true,
        "has_variants": false,
        "has_machine_quality": true,
        "preferred_quality_type": "normal",
        "machine_extruder_trains":
        {
            "0": "anycubic_i3_mega_s_extruder_0"
        }
    },

    "overrides":
    {
        "machine_name": { "default_value": "Anycubic i3 Mega (S Upgrade)" },
        "machine_heated_bed": { "default_value": true },
        "machine_width": { "default_value": 210 },
        "machine_height": { "default_value": 205 },
        "machine_depth": { "default_value": 210 },
        "machine_center_is_zero": { "default_value": false },
        "gantry_height": { "value": "0" },
        "machine_gcode_flavor": { "default_value": "RepRap (Marlin/Sprinter)" },
        "machine_start_gcode": { "default_value": "G21                                        ; metric values \nG90                                        ; absolute positioning \nM82                                        ; set extruder to absolute mode \nM107                                       ; start with the fan off \nM140 S{material_bed_temperature_layer_0}           ; Start heating the bed \nG4 S60                                     ; wait 1 minute \nM104 S{material_initial_print_temperature}         ; start heating the hot end \nM190 S{material_bed_temperature_layer_0}           ; wait for bed \nM109 S{material_initial_print_temperature}         ; wait for hotend \nM300 S1000 P500                            ; BEEP heating done \nG28 X0 Y10 Z0                              ; move X/Y to min endstops \nM420 S1                                    ; Enable leveling \nM420 Z2.0                                  ; Set leveling fading height to 2 mm \nG0 Z0.15                                   ; lift nozzle a bit \nG92 E0                                     ; zero the extruded length \nG1 X50 E25 F500                            ; Extrude 25mm of filament in a 5cm line. \nG92 E0                                     ; zero the extruded length again \nG1 E-2 F500                                ; Retract a little \nG1 X120 F4000                              ; Quickly wipe away from the filament line`" },
        "machine_end_gcode": { "default_value": "M104 S0                                    ; Extruder off \nM140 S0                                    ; Heatbed off \nM107                                       ; Fan off \nG91                                        ; relative positioning \nG1 E-5 F300                                ; retract a little \nG1 Z+10 E-5 X-20 Y-20 F{travel_xy_speed} ; lift print head \nG28 X0 Y0                                  ; homing \nG1 Y180 F2000                              ; reset feedrate \nM84                                        ; disable stepper motors \nG90                                        ; absolute positioning \nM300 S440 P200                             ; Make Print Completed Tones \nM300 S660 P250                             ; beep \nM300 S880 P300                             ; beep" },

        "machine_max_acceleration_x": { "value": 3000 },
        "machine_max_acceleration_y": { "value": 3000 },
        "machine_max_acceleration_z": { "value": 3000 },
        "machine_max_acceleration_e": { "value": 3000 },
        "machine_acceleration": { "value": 3000 },
        "machine_max_jerk_xy": { "value": 10 },
        "machine_max_jerk_z": { "value": 0.4 },
        "machine_max_jerk_e": { "value": 5 },
        "material_diameter": { "default_value": 1.75 },
        "acceleration_print": { "value": 1500 },
        "acceleration_travel": { "value": 3000 },
        "acceleration_travel_layer_0": { "value": "acceleration_travel" },
        "acceleration_roofing": { "enabled": "acceleration_enabled and roofing_layer_count > 0 and top_layers > 0" },

        "jerk_print": { "value": 7 },
        "jerk_travel": { "value": 8 },
        "jerk_travel_layer_0": { "value": "jerk_travel" },

        "acceleration_enabled": { "value": false },
        "jerk_enabled": { "value": true },

        "speed_print": { "value": 50.0 } ,
        "speed_infill": { "value": "speed_print" },
        "speed_wall": { "value": "speed_print / 2" },
        "speed_wall_0": { "value": "speed_wall" },
        "speed_wall_x": { "value": "speed_wall" },
        "speed_topbottom": { "value": "speed_print / 2" },
        "speed_roofing": { "value": "speed_topbottom" },
        "speed_travel": { "value": 100.0,  "maximum_value_warning": 150.0 },
        "speed_layer_0": { "value": 20.0 },
        "speed_print_layer_0": { "value": "speed_layer_0" },
[... 53 lines cut ...]
        "support_infill_rate": { "value": "0 if support_enable and support_structure == 'tree' else 20" },
        "support_use_towers": { "value": false },
        "support_xy_distance": { "value": "wall_line_width_0 * 2" },
        "support_xy_distance_overhang": { "value": "wall_line_width_0" },
        "support_z_distance": { "value": "layer_height if layer_height >= 0.16 else layer_height * 2" },
[... cut ...]
        "support_interface_height": { "value": "layer_height * 4" },
        "support_interface_density": { "value": 33.333 },
        "support_interface_pattern": { "value": "'grid'" },
        "support_interface_skip_height": { "value": 0.2 },
        "minimum_support_area": { "value": 2 },
        "minimum_interface_area": { "value": 10 },
        "top_bottom_thickness": { "value":  "layer_height_0 + layer_height * math.floor(1.2 / layer_height)" },
        "wall_thickness": { "value": "line_width * 3" }
    }
}

```

closed 2022-01-04 by Empire-of-E-Projects

Ghostkeeper, 2022-01-14:
Perhaps you'd like to create a pull request with your corrections?

I must say that it's unusual to print the first layer colder than the rest of the layers. Normally it's hotter. But if you tested it and it works, we'll believe you.

However you closed this issue. Is it not an issue any more?