Ultimaker/Cura #11198: Temperature tower is useless due to the different number of layers for each temperature

KravetsIgor, 2022-01-05:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ultimaker Cura

### Reproduction steps

1. Extentions->Part for calibration->Add PLA Temp Tower
2. Slice at Standard Q (0.2mm)

### Actual results

Layer 208 is redundant. 
200C bridge contains 4 layers. All other contains 3 layers. 

### Expected results

200C bridge must contain the same number of layers as others.
This makes temperature test useless.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No files needed.

Asterchades, 2022-01-05:
Given this is a problem with an extension, not the base client, it would be better (and actually useful) to report it to the extension's author. They may or may not be on the Cura team but they'll certainly be handling their bugs separately.

fvrmr, 2022-01-06:
Closing this issue since we don't maintain this plugin like @Asterchades mentioned.
You can report your bug here https://github.com/5axes/Calibration-Shapes/issues related to the calibration plugin.

closed 2022-01-06 by fvrmr