Ultimaker/Cura #11201: Error when opening marketplace

martijnschouten, 2022-01-05:
### Application Version

4.12.1

### Platform

windows 10

### Printer

Diabase H-series/ Ultimaker S5/ Ultimaker S2

### Reproduction steps

1. Install cura
2. Open cura
3. Press the marketplace button

### Actual results

I get the message: "Could not connect to the Cura Package Database. Please check your connection.

- I checked the fireway and cura is set as an exception
- I tried Test-NetConnection api.ultimaker.com 443. The result is TcpTestSucceeded = True
- Installed the latest updates in windows update
- I looked into the log file it says: got an QNetworkReplyError SSL handshake failed. The server returned: b''

### Expected results

It should just show me the marketplace.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7813642/cura.log)

fvrmr, 2022-01-07:
Hi @martijnschouten thank you for your report. 
This looks like an outdated SSL, see also: https://github.com/Ultimaker/Cura/issues/5494
Could you check if this suggestion: https://github.com/Ultimaker/Cura/issues/5494#issuecomment-999615224 fixes your issue?

martijnschouten, 2022-01-07:
Hi @fvrmr, Thank you for your reply. I have the latest updates installed in windows update. Is there anything else I could do to make sure I have the latest root certificates?

Ghostkeeper, 2022-01-14:
The updated SSL certificates were marked as a "functional update" (not a security update) for Windows so it could be that those are not installed. In addition, the group policy can prevent SSL root certificates from being installed.

Here is a myriad of other ways to install them: http://woshub.com/updating-trusted-root-certificates-in-windows-10/

Of course, the SSL handshake could also legitimately fail, i.e. because something is modifying your communication with the Marketplace. Some invasive firewalls are known to meddle with connections like this, which fails if they can't install a self-signed certificate. (Don't turn them off though unless you know what you're doing!)

If you want to check if you have the correct SSL certificates installed, look into `certmgr.exe`. You should see a certificate called ISRG Root X1 with the domain community.letsencrypt.org. You should not have DST Root CA X3, which is the old, expired certificate. Again, a Windows update would install it, but if they are not there then that's where it went wrong.

I'm not sure if we can help you further with this. We can't reproduce this issue with the testers. If the correct root certificate is present then I think we'll have to see if other people also have this and see what the commonalities between you two are to determine where the problem is.

closed 2022-01-31 by no-response[bot]