Ultimaker/Cura #11202: fehle beim laden einer stl datei aus den internnet

ppm1951, 2022-01-05:
### Application Version

4.12.1 und 4.13.1 Beta

### Platform

Windows 10

### Printer

anycubic i3 mega S

### Reproduction steps

Datei aus dem Internet geladen und die Fehlermeldung sie Screenshot.
[image]

ich kann kein slicing vornehmen oder daten eingeben/ verändern

wie löse ich das Problem

mfg
Martinetz

### Actual results

cua auch mehmals neu installiet und fehler noch vorhanden

### Expected results

wie kann ich das problem beheben

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2022-01-06:
Hi @ppm1951 thank you for your report. 
Looking at your screenshots you are loading a g-code file instead of a stl. 
When importing a g-code file you can't change the print settings. 
Let me know if that is the case. I will close this issue but feel free to reach out if you have more questions.

closed 2022-01-06 by fvrmr

ppm1951, 2022-01-06:
Danke fü ihre Hilfe 

 

Mit freundlichen grüßen 

Direktor

Hans Georg Martinetz 

Konstruktion Abteilung Germany 

Personnel Power & More AG Ltd.

International Transports

Vehicle Construction 

                         

 ***@***.***> ***@***.***             

Mobil +491702757982   Whats App

 

Diese Email einschließŸlich eventuell angehängter Dateien enthält

vertrauliche und/oder rechtlich geschützte Informationen.

Wenn Sie nicht der richtige Adressat sind und diese E-Mail irrtümlich erhalten haben, dürfen Sie weder den Inhalt der E-Mail

nutzen noch dürfen Sie eventuell angehängte Dateien Öffnen und auch nichts kopieren oder weitergeben.

Bitte verständigen Sie den Absender und löschen Sie diese

E-Mail. Vielen Dank.