Ultimaker/Cura #11203: Found bug on deleting filament types in other languages

ostfriese273, 2022-01-05:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Hello Ultimaker-Team
I discovered a bug on your German Settings. If you try to delete an filament type, you get asked if you really want to do this. Intead of the answer possibilities "Ja / Nein" their stand just "Yes / No". I know it is not a huge bug or problem nut it is a nice little detail you could fix. 

I would appreciate if you fix it
Hannes Lücht

[image]

### Actual results

cmake . && ctest --verbose

### Expected results

cmake . && ctest --verbose

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

cmake . && ctest --verbose

fvrmr, 2022-01-06:
Hi @ostfriese273 thank you for your report.
I have linked your issue to an existing ticket on our backlog related to some other translation bugs.

Ghostkeeper, 2022-01-14:
This one is a bit tricky.

The dialogue there is created with Yes and No buttons by adding `Qt.StandardButton` [flags to the dialogue](https://github.com/Ultimaker/Cura/blob/404e02d585ad637330fd98ef8b8f7e569551b376/resources/qml/Preferences/Materials/MaterialsPage.qml#L316).

These StandardButtons don't have any fields for [customisation](https://doc.qt.io/qt-5/qdialogbuttonbox.html#StandardButton-enum). We can just tell it which button to show, from a collection of standard buttons. And Qt links it up with the Escape key pressing cancel, the enter key pressing OK, etc.

Can they be translated? Apparently, yes, they get translated into the system language (which is also German for you as can be seen in the screenshot). However, that [requires the translation files](https://stackoverflow.com/a/4627177) of Qt to be present. People (in multiple forums) are saying you need to make sure to copy those files.

However those files are not provided in the PyQt package. You can [download the wheel](https://pypi.org/project/PyQt5/5.15.6/), open it up like a .zip file and peek inside. Those files are not there.

We could build Qt in our build environment and copy them (we already do build Qt for Linux and MacOS). But that also brings risk and maintenance.