Ultimaker/Cura #11204: Cura arachne engine beta v2 profile issues

ruredi87, 2022-01-05:
### Application Version

Latest arachne

### Platform

windows 11

### Printer

ender 3 v2

### Reproduction steps

[image]

I cannot import my profile. I saved it 5 seconds ago from arachne. Also Im trying to import arachne. Besides than I cannot save profile changes, It resets itself from first default values.

### Actual results

Profile changes does not save also imports failed.

### Expected results

Profile changes and importing profiles could be succesfull

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[project.zip](https://github.com/Ultimaker/Cura/files/7815738/project.zip)

ruredi87, 2022-01-05:
[cura deneme.zip](https://github.com/Ultimaker/Cura/files/7815743/cura.deneme.zip)
 this is the profile I am trying to import

ruredi87, 2022-01-05:
I found the issue. I have also cura 4.12,Arachne defaulty importing old cura profiles when first install, but after When we try to change or exporting, importing profiles error coming up. I deleted all profiles from arachne then manually copied one by one from cura 4.12.

closed 2022-01-05 by ruredi87

Ghostkeeper, 2022-01-14:
Due to the way Arachne was released, we only have an upgrade path from Cura 4.9 to Arachne. This will be fixed for the stable release.

We might want to improve that error message though...