Ultimaker/Cura #11205: Select edge in illistrator view (e.g. select side on view cube) selects additional edges when using edge only option

robbmart, 2022-01-06:
### Application Version

4.12.1

### Platform

Win 10.0.19043

### Printer

Ender 5 Plus

### Reproduction steps

1. Select Right, Left, Back, Front, Top, or Bottom view from view cube (found in top right corner). THIS IS A 2 DIMENSIONTAL VIEW!
2. Drag and select area
3. Toggle to "Edges Only" filter
4. EDGES  NOT PART OF THE 2 DIMENSIONAL VIEW BECOME SELECTED. 

edges invisible in selected view also become selected although invisible. This is annoying AF and creates SOOOO much more work... And I am talking about ANY eged not visible in the 2 dimension sketch view.

### Actual results

Edges not part of the sketch 2-dimensional view are also selected.

### Expected results

Only the edges visible to the sketch view should be selected. 

I can send a screen recording of what I am talking about

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2022-01-06:
Hi @robbmart thank you for your report.
Could you share a screen recording of what you mean?
Thank you in advance.

fieldOfView, 2022-01-06:
There is no "Edges only" filter in Cura, and neither is there a view cube in the top right corner. I think you are mixing up your applications.

closed 2022-01-20 by no-response[bot]