Ultimaker/Cura #11208: Odd network behavior

colinluthier, 2022-01-06:
### Application Version

4.8.0 and 4.12.1

### Platform

Windows 10

### Printer

CReality Ender-3 Pro

### Reproduction steps

1. Start the program 
2. Import Model
3. Slice

### Actual results

Ubiquity router crashes and the entire network fails. The router must be power cycled to restore the network.

### Expected results

Cura or one of it's plugins should not be able to crash a network. This has been happening for months, I only just realized it only happens on a fresh start of Cura. If I leave Cura open I have no problems after the first time rebooting the network.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7824075/cura.log)

fvrmr, 2022-01-07:
Hi @colinluthier thank you for your report.
Could you try to disable the one of both of these plugins: The UM3 networking plugin and or Octoprint plugin. 
Go to Marketplace - installed and disable these plugins.

I am not sure if this will work, so let me know!

colinluthier, 2022-01-07:
I tried to disable UM3 and it still crashes. I don't seem to have Octoprint installed. I have had it installed in older versions. I did a text search of the log for Octoprint and it came up blank.

colinluthier, 2022-01-07:
I have also tried starting Cura without USB detection and also without any community plugins.

colinluthier, 2022-01-08:
I believe I have this problem solved. I had a USB cable plugged in between the computer and the Ubiquity USG to have shell access and when Cura would look for attached printers it would crash the  USG. Interestingly I thought I had any USB checks shut off. I didn't need the cable, I only kept it there so I didn't lose it. I have it removed now and have been able to boot Cura without problems.

While thee is still something to repair here, I doubt the limited nature of the problem would warrant much time debugging.

fvrmr, 2022-01-11:
Thank you for replying, if the issue is with the USB cable you need to disable the USB printing plugin.
Read here more about this issue: https://support.ultimaker.com/hc/en-us/articles/4407742254610-Ultimaker-Cura-interacts-with-microcontrollers 
I will close this issue since this will solve your problem!

closed 2022-01-11 by fvrmr