Ultimaker/Cura #11209: odd artifacts during print causing a no material error during print

Payton-C3DM, 2022-01-06:
### Application Version

4.12.1

### Platform

windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. Entered the file to slice 
2. sliced and it produces the lines that I believe are causing the error
3. sliced again in ABS instead of PLA and said lines are mising

### Actual results

the lines shown in the images appear and when the printer is printing those layers it causes a no material error, but if you just continue the print it continues fine until it stops again. once it is past these lines it prints just fine. 

when printed in ABS the slicer did not show these lines and there was no errors during the print

### Expected results

a normal print with no errors on certain layers

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[New Compressed (zipped) Folder.zip](https://github.com/Ultimaker/Cura/files/7824531/New.Compressed.zipped.Folder.zip)

fvrmr, 2022-01-07:
Hi @Payton-C3DM thank you for your report.
Could you also share your project files (files - save project) instead of the .ufp files.

Ghostkeeper, 2022-01-14:
Those lines are Skin Edge Support. Skin Edge Support Layers can be set to 0, to remove them. They are intended to improve the overhangs of skin resting on the infill though. They tend to reduce pillowing and holes in the top of the print.

The lines themselves look good to me. Maybe the feeder is slipping a bit when it needs to make retractions? I'd try adjusting the feeders with the tightening bolt a bit.